elementary/settings-power #1: crashes switchboard when plug first opened

elementaryBot, 2017-04-21:
Clicking on the power plug in switchboard immediately crashes switchboard. This is the stacktrace: 

[FATAL 16:20:26.191107] [Gtk] gtk_widget_set_visible: assertion 'GTK_IS_WIDGET (widget)' failed
[FATAL 16:20:26.193852] power_upower_device_Refresh: assertion 'self != NULL' failed
[FATAL 16:20:26.193886] power_upower_device_get_Online: assertion 'self != NULL' failed
[FATAL 16:20:26.193994] [Gtk] gtk_stack_switcher_get_stack: assertion 'GTK_IS_STACK_SWITCHER (switcher)' failed
Segmentation fault (core dumped)

Launchpad Details: [#LP1577595](https://bugs.launchpad.net/bugs/1577595) Sam Thomas - 2016-05-02 23:23:03 +0000

elementaryBot, 2017-04-21:
it happens to me, but, I think it is deuplicate of this report
https://bugs.launchpad.net/switchboard-plug-power/+bug/1556607

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-power/+bug/1577595/comments/1) Jung-Kyu Park - 2016-05-04 18:39:34 +0000

elementaryBot, 2017-04-21:
i think this is not a duplicate of the bug you provide. here the error is about upower devices, dbus connections. the other is about gsettings schema's those are different.

@Sam are you running an official released build of elementary os?
if so could you provide the full output of the following command:

"switchboard -d power" 

thank you in advance

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-power/+bug/1577595/comments/2) kay van der Zander - 2016-05-24 17:41:44 +0000

elementaryBot, 2017-04-21:
I am running the official development version of Loki. I am running the latest version of the power plug which is 0.2.2+r306~daily~ubuntu0.4.1

Here's the output of "switchboard -d power"

[INFO 12:19:41.816708] Application.vala:155: System Settings version: 2.0
[INFO 12:19:41.816745] Application.vala:157: Kernel version: 4.4.0-22-generic
[DEBUG 12:19:41.817453] [Gtk] Connecting to session manager
[DEBUG 12:19:41.819391] DisplayPlug.vala:206: Activating Display plug
[DEBUG 12:19:41.819487] PlugsManager.vala:91: pantheon-display registered
[DEBUG 12:19:41.820086] ColorPlug.vala:92: Activating Color plug
[DEBUG 12:19:41.820156] PlugsManager.vala:91: hardware-gcc-color registered
[DEBUG 12:19:41.820654] SoundPlug.vala:142: Activating Sound plug
[DEBUG 12:19:41.820702] PlugsManager.vala:91: hardware-gcc-sound registered
[DEBUG 12:19:41.821203] WacomPlug.vala:92: Activating Wacom plug
[DEBUG 12:19:41.821252] PlugsManager.vala:91: hardware-gcc-wacom registered
[DEBUG 12:19:41.821851] keyboard.vala:52: Activating Keyboard plug
[DEBUG 12:19:41.821918] PlugsManager.vala:91: hardware-pantheon-keyboard registered
[DEBUG 12:19:41.822438] mouse-touchpad.vala:122: Activating Mouse-Touchpad plug
[DEBUG 12:19:41.822492] PlugsManager.vala:91: pantheon-mouse-touchpad registered
[DEBUG 12:19:41.823266] Plug.vala:381: Activating Power plug
[DEBUG 12:19:41.826017] Battery.vala:75: it is a dekstop (laptops false)
[DEBUG 12:19:41.827008] PowerSupply.vala:60: power supply path:  dbus connected

** (process:1589): CRITICAL **: string_contains: assertion 'self != NULL' failed

** (process:1589): CRITICAL **: string_contains: assertion 'self != NULL' failed
[DEBUG 12:19:41.834788] CliCommunicator.vala:100: lid_close_dock:ignore
[DEBUG 12:19:41.834829] CliCommunicator.vala:105: lid_close:suspend
[DEBUG 12:19:41.849780] PlugsManager.vala:91: system-pantheon-power registered
[DEBUG 12:19:41.855890] Plug.vala:199: Activating Printers plug
[DEBUG 12:19:41.855979] PlugsManager.vala:91: pantheon-printers registered
[DEBUG 12:19:41.856627] BluetoothPlug.vala:103: Activating Bluetooth plug
[DEBUG 12:19:41.856678] PlugsManager.vala:91: network-gcc-bluetooth registered
[DEBUG 12:19:41.858020] Plug.vala:207: Activating Network plug
[DEBUG 12:19:41.908916] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy'
[DEBUG 12:19:41.909262] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy.ftp'
[DEBUG 12:19:41.909403] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy.http'
[DEBUG 12:19:41.909536] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy.https'
[DEBUG 12:19:41.909677] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy.socks'
[DEBUG 12:19:41.909824] PlugsManager.vala:91: pantheon-network registered
[DEBUG 12:19:41.929042] OnlineAccounts.vala:290: Activating Online Accounts plug
[DEBUG 12:19:41.929903] microsoft.vala:73: OnlineAccouts: Activating Microsoft plugin
[DEBUG 12:19:41.930238] google.vala:94: OnlineAccouts: Activating Google plugin
[DEBUG 12:19:41.930537] facebook.vala:58: OnlineAccouts: Activating Facebook plugin
[DEBUG 12:19:41.930831] yahoo.vala:85: OnlineAccouts: Activating Yahoo plugin
[DEBUG 12:19:41.931136] LastFM.vala:41: OnlineAccouts: Activating LastFM plugin
[DEBUG 12:19:41.931327] fastmail.vala:39: OnlineAccouts: Activating FastMail plugin
[DEBUG 12:19:41.931579] OAuth.vala:41: OnlineAccouts: Activating Generic OAuth plugin
[DEBUG 12:19:41.931815] Password.vala:41: OnlineAccouts: Activating Password plugin
[DEBUG 12:19:41.933608] PlugsManager.vala:91: network-pantheon-online-accounts registered
[DEBUG 12:19:41.934303] Plug.vala:79: Activating Sharing plug
[DEBUG 12:19:41.934371] PlugsManager.vala:91: pantheon-sharing registered
[DEBUG 12:19:41.934951] Plug.vala:109: Activating Applications plug
[DEBUG 12:19:41.935316] PlugsManager.vala:91: personal-pantheon-applications registered
[DEBUG 12:19:41.936513] desktop-plug.vala:178: Activating Desktop plug
[DEBUG 12:19:41.936579] PlugsManager.vala:91: pantheon-desktop registered
[DEBUG 12:19:41.937451] Plug.vala:193: Activating Locale plug
[DEBUG 12:19:41.937523] PlugsManager.vala:91: system-pantheon-locale registered
[DEBUG 12:19:41.938144] Plug.vala:91: Activating Notifications plug
[DEBUG 12:19:41.938212] PlugsManager.vala:91: personal-pantheon-notifications registered
[DEBUG 12:19:41.938851] Plug.vala:127: Activating Security & Privacy plug
[DEBUG 12:19:41.938915] PlugsManager.vala:91: pantheon-security-privacy registered
[DEBUG 12:19:41.946988] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.interface'
[DEBUG 12:19:41.982755] Settings.vala:163: Loading settings from schema 'org.pantheon.desktop.gala.appearance'
[DEBUG 12:19:41.983455] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.wm.preferences'
[DEBUG 12:19:41.991968] PlugsManager.vala:91: pantheon-tweaks registered
[INFO 12:19:41.992840] Tweaks.vala:229: Activating Tweak plug
[DEBUG 12:19:41.997725] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.wm.preferences'
[DEBUG 12:19:41.999811] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.interface'
[DEBUG 12:19:42.010544] Settings.vala:163: Loading settings from schema 'org.pantheon.desktop.gala.appearance'
[... 169 lines cut ...]
[DEBUG 12:19:42.152591] PlugsSearch.vala:57: plugsSearch: add open window:  
[... cut ...]
[FATAL 12:19:42.212771] [Gtk] gtk_widget_set_visible: assertion 'GTK_IS_WIDGET (widget)' failed
[FATAL 12:19:42.217217] power_upower_device_Refresh: assertion 'self != NULL' failed
[FATAL 12:19:42.217256] power_upower_device_get_Online: assertion 'self != NULL' failed
[FATAL 12:19:42.217297] [Gtk] gtk_stack_switcher_get_stack: assertion 'GTK_IS_STACK_SWITCHER (switcher)' failed
Segmentation fault (core dumped)

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-power/+bug/1577595/comments/3) Sam Thomas - 2016-05-24 19:20:56 +0000

elementaryBot, 2017-04-21:
[Expired for Switchboard Power Plug because there has been no activity for 60 days.]

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-power/+bug/1577595/comments/4) Launchpad Janitor - 2016-07-24 04:17:34 +0000

elementaryBot, 2017-04-21:
Marking this as "new" since the reporter provided extra information.

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-power/+bug/1577595/comments/5) Zisu Andrei - 2016-07-25 07:57:20 +0000

naaando, 2017-04-27:
It was closed on [this merge](https://code.launchpad.net/~kay20/switchboard-plug-power/fix-1577595/+merge/310007) ?

closed 2017-04-27 by danrabbit