elementary/settings-power #15: Brightness is not adjusted automatically

elementaryBot, 2017-04-21:
Brightness is not adjusted automatically when I enable the corresponding option.
Macbook pro 5,3, nvidia graphics card, 0.4 beta.

Launchpad Details: [#LP1592955](https://bugs.launchpad.net/bugs/1592955) Ioannis Lazaridis - 2016-06-15 19:33:37 +0000

elementaryBot, 2017-04-21:
please run "switchboard -d power" and provide the output please

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-power/+bug/1592955/comments/1) kay van der Zander - 2016-07-15 11:17:05 +0000

elementaryBot, 2017-04-21:
I'm unsure how helpful my log is:

matzipan@matzipan-thinkpad:~$ switchboard -d power
[INFO 10:18:47.086919] Application.vala:155: System Settings version: 2.0
[INFO 10:18:47.086969] Application.vala:157: Kernel version: 4.6.0-040600rc7-generic
[DEBUG 10:18:47.087842] [Gtk] Connecting to session manager
[DEBUG 10:18:47.090174] DisplayPlug.vala:206: Activating Display plug
[DEBUG 10:18:47.090305] PlugsManager.vala:91: pantheon-display registered
[DEBUG 10:18:47.090953] ColorPlug.vala:92: Activating Color plug
[DEBUG 10:18:47.091058] PlugsManager.vala:91: hardware-gcc-color registered
[DEBUG 10:18:47.091652] SoundPlug.vala:142: Activating Sound plug
[DEBUG 10:18:47.091731] PlugsManager.vala:91: hardware-gcc-sound registered
[DEBUG 10:18:47.092260] WacomPlug.vala:92: Activating Wacom plug
[DEBUG 10:18:47.092339] PlugsManager.vala:91: hardware-gcc-wacom registered
[DEBUG 10:18:47.092958] keyboard.vala:51: Activating Keyboard plug
[DEBUG 10:18:47.093055] PlugsManager.vala:91: hardware-pantheon-keyboard registered
[DEBUG 10:18:47.093609] mouse-touchpad.vala:122: Activating Mouse-Touchpad plug
[DEBUG 10:18:47.093699] PlugsManager.vala:91: pantheon-mouse-touchpad registered
[DEBUG 10:18:47.094510] Plug.vala:388: Activating Power plug
[DEBUG 10:18:47.102262] Battery.vala:69: battery path:/org/freedesktop/UPower/devices/battery_BAT0 , its a laptop, dbus connected
[DEBUG 10:18:47.105196] PowerSupply.vala:58: power supply path: (null) dbus connected

** (process:7015): CRITICAL **: string_contains: assertion 'self != NULL' failed

** (process:7015): CRITICAL **: string_contains: assertion 'self != NULL' failed
[DEBUG 10:18:47.112785] CliCommunicator.vala:102: lid_close_dock:ignore
[DEBUG 10:18:47.112846] CliCommunicator.vala:107: lid_close:suspend
[DEBUG 10:18:47.127033] PlugsManager.vala:91: system-pantheon-power registered
[DEBUG 10:18:47.133862] Plug.vala:199: Activating Printers plug
[DEBUG 10:18:47.133978] PlugsManager.vala:91: pantheon-printers registered
[DEBUG 10:18:47.134721] BluetoothPlug.vala:103: Activating Bluetooth plug
[DEBUG 10:18:47.134795] PlugsManager.vala:91: network-gcc-bluetooth registered
[DEBUG 10:18:47.136233] Plug.vala:207: Activating Network plug
[DEBUG 10:18:47.174588] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy'
[DEBUG 10:18:47.175174] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy.ftp'
[DEBUG 10:18:47.175392] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy.http'
[DEBUG 10:18:47.175606] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy.https'
[DEBUG 10:18:47.175806] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy.socks'
[DEBUG 10:18:47.176003] PlugsManager.vala:91: pantheon-network registered
[DEBUG 10:18:47.195607] OnlineAccounts.vala:290: Activating Online Accounts plug
[DEBUG 10:18:47.196131] Password.vala:41: OnlineAccouts: Activating Password plugin
[DEBUG 10:18:47.196653] yahoo.vala:85: OnlineAccouts: Activating Yahoo plugin
[DEBUG 10:18:47.197006] facebook.vala:58: OnlineAccouts: Activating Facebook plugin
[DEBUG 10:18:47.197337] microsoft.vala:73: OnlineAccouts: Activating Microsoft plugin
[DEBUG 10:18:47.197662] google.vala:94: OnlineAccouts: Activating Google plugin
[DEBUG 10:18:47.198000] LastFM.vala:41: OnlineAccouts: Activating LastFM plugin
[DEBUG 10:18:47.198216] fastmail.vala:39: OnlineAccouts: Activating FastMail plugin
[DEBUG 10:18:47.198499] OAuth.vala:41: OnlineAccouts: Activating Generic OAuth plugin
[DEBUG 10:18:47.201085] PlugsManager.vala:91: network-pantheon-online-accounts registered
[DEBUG 10:18:47.201891] Plug.vala:123: Activating Sharing plug
[DEBUG 10:18:47.201984] PlugsManager.vala:91: pantheon-sharing registered
[DEBUG 10:18:47.202720] Plug.vala:109: Activating Applications plug
[DEBUG 10:18:47.202840] PlugsManager.vala:91: personal-pantheon-applications registered
[DEBUG 10:18:47.204146] desktop-plug.vala:178: Activating Desktop plug
[DEBUG 10:18:47.204260] PlugsManager.vala:91: pantheon-desktop registered
[DEBUG 10:18:47.205195] Plug.vala:193: Activating Locale plug
[DEBUG 10:18:47.205295] PlugsManager.vala:91: system-pantheon-locale registered
[DEBUG 10:18:47.205905] Plug.vala:91: Activating Notifications plug
[DEBUG 10:18:47.206004] PlugsManager.vala:91: personal-pantheon-notifications registered
[DEBUG 10:18:47.206668] Plug.vala:127: Activating Security & Privacy plug
[DEBUG 10:18:47.206767] PlugsManager.vala:91: pantheon-security-privacy registered
[DEBUG 10:18:47.207809] Plug.vala:198: Activating Parental Controls plug
[DEBUG 10:18:47.207906] PlugsManager.vala:91: pantheon-parental-controls registered
[DEBUG 10:18:47.208906] Plug.vala:559: Activating About plug
[DEBUG 10:18:47.209008] PlugsManager.vala:91: system-pantheon-about registered
[DEBUG 10:18:47.210084] Plug.vala:137: Activating Accessibility plug
[DEBUG 10:18:47.210358] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.interface'
[DEBUG 10:18:47.210740] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.a11y.keyboard'
[DEBUG 10:18:47.211189] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.a11y'
[DEBUG 10:18:47.211345] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.a11y.magnifier'
[... 187 lines cut ...]
[FATAL 10:18:47.377413] [GLib] g_sequence_get: assertion '!is_end (iter)' failed
[FATAL 10:18:47.378545] [GLib] g_sequence_get: assertion '!is_end (iter)' failed
[... cut ...]
[FATAL 10:18:47.392633] [GLib] g_sequence_get: assertion '!is_end (iter)' failed
[FATAL 10:18:47.394481] [GLib] g_sequence_get: assertion '!is_end (iter)' failed
[FATAL 10:18:47.396559] [GLib] g_sequence_get: assertion '!is_end (iter)' failed
[FATAL 10:18:47.398638] [GLib] g_sequence_get: assertion '!is_end (iter)' failed
[FATAL 10:18:47.401297] [GLib] g_sequence_get: assertion '!is_end (iter)' failed

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-power/+bug/1592955/comments/2) Zisu Andrei - 2016-07-18 08:19:29 +0000

4jNsY6fCVqZv, 2019-08-10:
What exactly does this setting do? I have activated it. The other time it was deactivated for testing. I didn't notice any changes.