elementary/settings-power #16: System Settings a bit slow to load.

elementaryBot, 2017-04-21:
When user clicks System Settings, there is a 7-10s delay between when the user clicks and when it shows up on the Desktop.  All other applications load instantly.

Machine is using an Intel 750 Pro 400GB NVMe PCI-e Drive.

Launchpad Details: [#LP1622193](https://bugs.launchpad.net/bugs/1622193) Mikel Jagan - 2016-09-10 14:37:25 +0000

elementaryBot, 2017-04-21:
OP, can you list anything you've done since installing like additional software you may have installed or software sources added?

Can you also open Terminal and run "switchboard -d" and then copy and paste the output here?

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-power/+bug/1622193/comments/1) Daniel Fore - 2016-09-10 21:33:26 +0000

elementaryBot, 2017-04-21:
switchboard -d
[INFO 18:30:27.530138] Application.vala:155: System Settings version: 2.0
[INFO 18:30:27.530200] Application.vala:157: Kernel version: 4.4.0-36-generic
[DEBUG 18:30:27.531689] [Gtk] Connecting to session manager
[DEBUG 18:30:27.535120] DisplayPlug.vala:211: Activating Display plug
[DEBUG 18:30:27.535267] PlugsManager.vala:91: pantheon-display registered
[DEBUG 18:30:27.536328] ColorPlug.vala:92: Activating Color plug
[DEBUG 18:30:27.536460] PlugsManager.vala:91: hardware-gcc-color registered
[DEBUG 18:30:27.537304] SoundPlug.vala:142: Activating Sound plug
[DEBUG 18:30:27.537390] PlugsManager.vala:91: hardware-gcc-sound registered
[DEBUG 18:30:27.538219] WacomPlug.vala:92: Activating Wacom plug
[DEBUG 18:30:27.538306] PlugsManager.vala:91: hardware-gcc-wacom registered
[DEBUG 18:30:27.539242] keyboard.vala:51: Activating Keyboard plug
[DEBUG 18:30:27.539346] PlugsManager.vala:91: hardware-pantheon-keyboard registered
[DEBUG 18:30:27.540237] mouse-touchpad.vala:122: Activating Mouse-Touchpad plug
[DEBUG 18:30:27.540340] PlugsManager.vala:91: pantheon-mouse-touchpad registered
[DEBUG 18:30:27.541491] Plug.vala:388: Activating Power plug
[DEBUG 18:30:28.389539] Battery.vala:72: it is a dekstop (laptops false)
[DEBUG 18:30:30.386264] PowerSupply.vala:58: power supply path: (null) dbus connected

** (process:7958): CRITICAL **: string_contains: assertion 'self != NULL' failed

** (process:7958): CRITICAL **: string_contains: assertion 'self != NULL' failed
[DEBUG 18:30:30.411412] CliCommunicator.vala:102: lid_close_dock:ignore
[DEBUG 18:30:30.411489] CliCommunicator.vala:107: lid_close:suspend
[DEBUG 18:30:30.420538] PlugsManager.vala:91: system-pantheon-power registered
[DEBUG 18:30:30.432507] Plug.vala:199: Activating Printers plug
[DEBUG 18:30:30.432667] PlugsManager.vala:91: pantheon-printers registered
[DEBUG 18:30:30.433955] BluetoothPlug.vala:103: Activating Bluetooth plug
[DEBUG 18:30:30.434075] PlugsManager.vala:91: network-gcc-bluetooth registered
[DEBUG 18:30:30.437057] Plug.vala:207: Activating Network plug
[DEBUG 18:30:30.485503] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy'
[DEBUG 18:30:30.486143] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy.ftp'
[DEBUG 18:30:30.486484] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy.http'
[DEBUG 18:30:30.486871] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy.https'
[DEBUG 18:30:30.487215] Settings.vala:163: Loading settings from schema 'org.gnome.system.proxy.socks'
[DEBUG 18:30:30.487560] PlugsManager.vala:91: pantheon-network registered
[DEBUG 18:30:30.524796] OnlineAccounts.vala:290: Activating Online Accounts plug
[DEBUG 18:30:30.525532] LastFM.vala:41: OnlineAccouts: Activating LastFM plugin
[DEBUG 18:30:30.526107] Email.vala:42: OnlineAccouts: Activating Mail plugin
[DEBUG 18:30:30.526626] fastmail.vala:42: OnlineAccouts: Activating FastMail plugin
[DEBUG 18:30:30.527096] Password.vala:41: OnlineAccouts: Activating Password plugin
[DEBUG 18:30:30.528471] PlugsManager.vala:91: network-pantheon-online-accounts registered
[DEBUG 18:30:30.529611] Plug.vala:123: Activating Sharing plug
[DEBUG 18:30:30.529727] PlugsManager.vala:91: pantheon-sharing registered
[DEBUG 18:30:30.530805] Plug.vala:109: Activating Applications plug
[DEBUG 18:30:30.530967] PlugsManager.vala:91: personal-pantheon-applications registered
[DEBUG 18:30:30.533097] desktop-plug.vala:178: Activating Desktop plug
[DEBUG 18:30:30.533234] PlugsManager.vala:91: pantheon-desktop registered
[DEBUG 18:30:30.534703] Plug.vala:193: Activating Locale plug
[DEBUG 18:30:30.534816] PlugsManager.vala:91: system-pantheon-locale registered
[DEBUG 18:30:30.535796] Plug.vala:91: Activating Notifications plug
[DEBUG 18:30:30.535917] PlugsManager.vala:91: personal-pantheon-notifications registered
[INFO 18:30:30.537054] Tweaks.vala:68: Activating tweaks plug
[DEBUG 18:30:30.537181] PlugsManager.vala:91: elementary-tweaks registered
[DEBUG 18:30:30.538242] Plug.vala:522: Activating About plug
[DEBUG 18:30:30.538362] PlugsManager.vala:91: system-pantheon-about registered
[DEBUG 18:30:30.539418] Plug.vala:137: Activating Accessibility plug
[DEBUG 18:30:30.539678] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.interface'
[DEBUG 18:30:30.540051] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.a11y.keyboard'
[DEBUG 18:30:30.540619] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.a11y'
[DEBUG 18:30:30.540845] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.a11y.magnifier'
[DEBUG 18:30:30.541482] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.a11y.applications'
[DEBUG 18:30:30.541728] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.wm.preferences'
[DEBUG 18:30:30.541954] Settings.vala:163: Loading settings from schema 'org.gnome.settings-daemon.peripherals.mouse'
[DEBUG 18:30:30.542195] Settings.vala:163: Loading settings from schema 'org.gnome.desktop.a11y.mouse'
[DEBUG 18:30:30.542560] Settings.vala:163: Loading settings from schema 'org.gnome.settings-daemon.plugins.media-keys'
[DEBUG 18:30:30.542754] Settings.vala:163: Loading settings from schema 'org.pantheon.desktop.gala.animations'
[DEBUG 18:30:30.542915] PlugsManager.vala:91: pantheon-accessibility registered
[DEBUG 18:30:30.544321] Plug.vala:288: Activating Date & Time plug
[... 189 lines cut ...]
[... cut ...]
[FATAL 18:30:30.648209] [GLib] g_sequence_get: assertion '!is_end (iter)' failed
[FATAL 18:30:30.650466] [GLib] g_sequence_get: assertion '!is_end (iter)' failed
[FATAL 18:30:30.652698] [GLib] g_sequence_get: assertion '!is_end (iter)' failed
[FATAL 18:30:30.655096] [GLib] g_sequence_get: assertion '!is_end (iter)' failed
[FATAL 18:30:30.657571] [GLib] g_sequence_get: assertion '!is_end (iter)' failed

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-power/+bug/1622193/comments/2) Mikel Jagan - 2016-09-10 22:31:12 +0000

elementaryBot, 2017-04-21:
Installed from AppCenter:
Chromium, Gimp, Gnome System Monitor

The following was executed in terminal after a fresh installation:
sudo apt-get install software-properties-common
sudo apt-get install ppa-purge
sudo add-apt-repository ppa:graphics-drivers
sudo add-apt-repository ppa:philip.scott/elementary-tweaks
sudo add-apt-repository ppa:snwh/pulp #Paper Icons and Theme

sudo apt-get update

sudo apt-get install nvidia-367 elementary-tweaks paper-icon-theme rar ubuntu-restricted-extras

sudo apt-get dist-upgrade -y

#Removed parental controls due to wingpanel freezing for five minutes -- wingpanel still froze.
sudo apt-get remove switchboard-plug-parental-controls

The following was removed with AppCenter:
Camera, Music, Calendar

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-power/+bug/1622193/comments/3) Mikel Jagan - 2016-09-10 22:42:59 +0000

elementaryBot, 2017-04-21:
The following was added from third party downloads:
Black Magic Fusion 8.2
3-D Coat 4.7
MODO 801 and MODO 10.0 v2
Substance Designer 5.5
Substance Painter 2
Unreal Engine 4.12.5
Perforce (P4V and P4Admin)
Octane Render 3.02
Krita
Spotify

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-power/+bug/1622193/comments/4) Mikel Jagan - 2016-09-10 22:46:21 +0000

elementaryBot, 2017-04-21:
System Specs:
Elementary OS 0.4 (Loki)
x2 Xeon E5-2696 v3
128GB DDR4 ECC RAM
x3 GeForce 980Ti SC
Intel 750 Series 400GB NVMe PCI-e flash drive (Elementary OS)
Samsung 950 Pro 512GB NVMe M.2 flash drive (Windows 10)

Let me know if you need any more information.

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-power/+bug/1622193/comments/5) Mikel Jagan - 2016-09-10 22:57:50 +0000

elementaryBot, 2017-04-21:
Looks like there's a 3 second pause once it hits the power plug. Re-assigning 

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-power/+bug/1622193/comments/6) Daniel Fore - 2016-09-11 01:01:40 +0000

kvdz, 2017-06-12:
that is not a pause, it makes connection with a Dbus interface, This takes time to be scheduled and wake from his sleep state.  see line: PowerSupply.vala:58: power supply path: (null) dbus connected 

what dbus does is make connection via IPC (inner process communication) with an other process.
This can't be speeded up it has todo with dbus itself.  The same counts for the 1 second you see on the battery check. there are other processes called who have this information