elementary/settings-power #21: Cannot set "Shutdown" for the power button

elementaryBot, 2017-04-21:
In system-settings I cannot set the power button to "Shutdown" instead I only get the options of "Sleep", "Hibernate", and "Do Nothing".

The "Shutdown" function needs to be put back in/enabled. This is for the sake of hardware.

I work at a computer shop and the a lot of our problems stem from Windows having the power button set to Sleep or Hibernate the computer. On laptops this drains the battery. On desktops it causes hard drive lock and boot issues is the power goes out while the system is in sleep or hibernation mode. We end up having to delete the "hiberfile.sys" and then run a CHKDSK. We always set the power button to shutdown the computer instead of the alternatives.

If you can "Shutdown" using the GUI then you should also be able to do so using the power button.

Here is what happened to myself recently and why I am filing a bug report. 

I used the power button to shutdown my computer as I was moving it. Instead of shutting down the system, it went into sleep mode. I Later unplugged my computer. When I set my computer back up, I could not access my separate NTFS drive.

I got the message that my drive had closed in an unsafe state and I should run CHKDSK. This left me to borrow my friends Windows computer and run a CHKDSK on my drive before I could use it again in my Linux machine.

For people who do not have access to a Windows machine, they would be boned.

Please enable the option to actually shutdown your computer using the power button.

Launchpad Details: [#LP1677065](https://bugs.launchpad.net/bugs/1677065) Eosdude - 2017-03-28 23:51:41 +0000

fchris82, 2017-06-20:
I don't believe it... This feature is not so big, and I think it is not so much work to fix it. But very-very annoying. There are many repeated post:
- https://elementaryos.stackexchange.com/questions/11358/missing-options-in-system-power-manager
- https://elementaryos.stackexchange.com/questions/12451/how-to-i-get-the-power-button-to-shut-down-the-computer
- https://elementaryos.stackexchange.com/questions/8849/how-can-i-configure-the-power-button-to-shutdown-the-computer (2016!!!)
Now it is 2017 summer, and nothing happened. But there are new battery icons: https://medium.com/elementaryos/loki-updates-for-may-1b44b4e942e3 ???

Could someone also tell me why it has been removed? I don't understand. :-/

DDieter, 2017-06-20:
Unfortunately this option was removed by the GNOME developers because they believe this action is "too destructive". Because this plug uses the GNOME settings to set the power button action, it's not possible to set the action to shutdown anymore.

If the shutdown action gets added again (which might happen, see this [issue](https://bugzilla.gnome.org/show_bug.cgi?id=781108)) then this would be possible again, or maybe we can find a workaround.

cyberplant, 2017-07-19:
Hello everybody!

PLEASE, we need a workaround. One of the things I like of the linux world is that we don't have to follow what daddy (Apple/MS) says.. I'm also a Mac and Android user and HATE when they remove features that were cool because they think nobody uses, or whatever reason!

In this case, what they did is idiotic, and we need to fix that somehow.

Now.. how should we proceed?

donadigo, 2017-07-19:
@cyberplant: one and the easiest solution would be to just add a distro patch with this functionality added back.

danrabbit, 2019-06-08:
I think this is probably covered by #111. Closing as fixed

closed 2019-06-08 by danrabbit