elementary/settings-power #39: Option to disable "suspend" at all.

cyberplant, 2017-07-19:
In my case, suspend doesn't work on this desktop computer; when my machine enters in suspend mode, it doesn't stop much of the hardware (disks, all fans, etc remains on) and also it never wakes again.. 

So I want to be able to remove/disable the suspend option from the menu, cause I sometimes I'm hurried and press this button instead of "shut down".

Is it possible to add a global variable on this panel to disable it? What do you think about that?

kvdz, 2017-07-20:
please not you wrote a bug report for the settings plug power,  not the indicator (top right)
this project is only that part in the  settings -> power.

danrabbit, 2017-07-20:
Closing as "invalid" as this report seems to be about https://github.com/elementary/wingpanel-indicator-session and not this repo

closed 2017-07-20 by danrabbit

cyberplant, 2017-07-21:
I expected to have this setting on the power settings on the "system settings" app..

After we settle on the setting, we can focus on the other part.. or am I missing something?