elementary/settings-power #45: After being suspended, the touchpad gets crazy

febrezo, 2017-10-17:
Once you are trying to recover from a suspension, the touchpad gets crazy and you can no longer use it. The mouse seems to work. This behaviour has been taking place for a few days even in the login screen that appears after being suspended (i. e., before login in). How can I help to know what Is happening? I have tried restarting the Windows Manager and so on but I'm making blind sticks.

febrezo, 2017-10-17:
I can confirm that this is happening only on one of my computers running elementary (a Slimbook Katana). I'm checking with the hardware provider if this is a known issue linked to the touchpad firmware.

eljefuri, 2018-11-13:
I am also having this problem on a Lenovo Ideapad. 1 out of 3 times when awaken from suspend, the trackpad is simply dead. The combination of this issue along with #144 means that, for a non-technical user, the machine must be REBOOTED every time it is opened. Hopefully this is obscure, but if not it sure seems like a very high priority issue.