elementary/settings-power #51: Detect hardware, not form factor

danrabbit, 2018-01-16:
https://github.com/elementary/switchboard-plug-power/blob/cf8fd1ca2c68a0ca6d2fdc966957f3994808250e/src/Plug.vala#L235

Instead of trying to find a laptop here we should be smarter and look for backlight, lid, battery etc separately. This would be more robust and support more form factors

augustin-b, 2018-01-20:
Do you have any idea how to done it?
Shoud we detect for all these things in one bash script or in separate ones or?

I currently made little backlight check and implemented it same way as laptop_detect currently is.

augustin-b, 2018-01-20:
Check this please:

[./switchboard-plug-power/src/Plug.vala](https://gist.github.com/augustin-b/f625962c66ba5c122da2e742062a73b5)

Build Test:

[./switchboard-plug-power/build$ sudo make install](https://gist.github.com/augustin-b/eeab3b15f9f675e5df685ccfba406a98)

danrabbit, 2018-01-21:
Hey @augustin-b thanks for looking into solving this issue! Can you do me a huge favor and create branches with pull requests instead of using a gist? That would really help to see what exactly your changes are

augustin-b, 2018-01-21:
Np, I'll do it.. but first it still needs some work...

PR #56 (2018-01-23): Battery detection
Fixes #49 
Fixes #51 

- method to detect battery
- if system has battery display proper settings
- laptop-detect script & laptop_detect removed
- handle GLib2.errors
- better indentation
closed 2018-01-23 by danrabbit