elementary/settings-power #78: Remove Hibernate options

cassidyjames, 2018-03-16:
These aren't doing anything for me; only suspend seems to work out of the two. I'm not sure hibernating makes a ton of sense outside of something smarter like hybrid suspend anyway since boot is relatively fast.

avivbarel, 2018-11-20:
Hibernating is useful for me, since I travel a lot, and when my computer is suspended it uses a lot of battery, I bet there others just like me.
A fix for the options would be better, it's not working for me, and according to google not only me.

cassidyjames, 2018-11-20:
@avivbarel right, ideally hibernating would work. But it doesn't, and I believe it's disabled in the Ubuntu kernel. Barring it actually working, we should not have nonfunctional options in the settings.

rubydesign, 2019-01-30:
@cassidyjames while i agree with the logic, i would really see hibernate working.
Like avivbarel my computer draws far too much power for suspend to be useful. It's basically the same as just leaving it on. And booting is by no means fast enough.

I have tried to get hibernate to work, but failed. I think this would be a real bonus for elementary. But this may not be the place, so where would the "get hibernation to work" issue go?

lewisgoddard, 2019-05-01:
I would much prefer to see improvements with hibernation instead of it being removed, although that would make sense in the mean time.

| Cons | Pros |
|-------|------|
| Hibernate doesn't work reliably. | Hibernate doesn't use power like suspend does, which leads to improved battery life. |
| | Hibernate locks disk encryption which I believe is now the default on elementary OS. |
| | Hibernate doesn't keep RAM active, keeping private details out of memory. |

IamLizu, 2019-05-03:
Whats the update on this? Doesn't seem to work my system either.

PR #112 (2019-05-07): Remove broken hibernate option
Fixes #78 unless/until hibernation is fixed/enabled in the OS.
closed 2019-05-07 by danrabbit