elementary/settings-power #88: Playing a game with joypad inputs (but no keyboard/mouse) causes the system to sleep after some time

alexgleason, 2018-06-25:
I have "Turn off display when inactive for" set to 15 minutes. Normally this works great. However, I started playing a game in VisualBoyAdvance, and after 15 minutes of play the screen shuts off. This is despite the fact that I am constantly using a joypad throughout this time.

I suggest that the setting take joypad input into account when determining how long the system has been "inactive".

quequotion, 2018-07-01:
I've had the same problem for as long as I have been using computers that could sleep and have game controllers connected to them, regardless of operating system.

On the one hand, it should probably be resolved by having the game software prevent the computer from entering sleep mode (like a video player would); but on the other I never understood why joypad inputs are ignored in the first place.

lenemter, 2024-02-10:
We don't handle any joypad inputs. I think libmutter or libinput should be responsible for this. Closing as out of scope.

closed 2024-02-10 by lenemter