elementary/settings-power #89: Never suspend when plugged in setting ignored (always uses battery setting for suspend)

ianmjones, 2018-07-05:
On laptop, I have "Suspend when inactive for:" set to "Never" for "Plugged In".

[image]

And for battery I have the same setting set to "30 min".

[image]

Even though my laptop is plugged in, if I leave it alone for 30 minutes then it suspends.

It seems like the "Plugged In" settings are being ignored.

Clean install of Juno Beta 1, updates applied. No manual manipulation of config. Have restarted a couple of times etc. and checked that settings still look as intended when revisiting the settings panel (i.e. they saved).

Was not an issue when previously using Loki.

If you need me to inspect any logs etc please let me know, happy to grab whatever you need (comfortable with command line).

crmlt, 2018-07-06:
I can confirm. I experience this too. All settings are completely ignored not only time to suspend.

danrabbit, 2018-07-25:
As far as I can tell, we're setting the correct settings for the GSD power plugin. This might be an issue with GSD

kyentei, 2018-09-05:
> Always uses battery settings for suspend
Actually, it just suspends after 30 minutes, regardless of battery setting. This is still a problem.

croozey, 2018-10-09:
This is STILL a **HUGE** issue for me. 3 out of 3 test boxes I have running, with the latest version of Juno Beta 2 are still suspending, despite the power setting being set to "Never" suspend when inactive. Once the PCs suspends, all containers and VMs obviously also suspend. For me this is a **show-stopper**. I can not upgrade my main machines until this is resolved. It is easy to test. Just load the lastest version of Juno on any old PC, change the power settings to "Never" suspend when inactive and wait... **PLEASE HELP** - I'd like to use Juno as my main OS, but at this rate this bug is going to slip through to the released version...

croozey, 2018-10-09:
If I can help in any way with testing, please each out to me... I would be more than happy to do what I can to get this bug resolved.

danrabbit, 2018-10-09:
@croozey This has been fixed in master. You'll need to wait for an update to be pushed. This is why we don't recommend using the beta on production machines. It is not a stable release and we do not recommend using it as your main OS. If you need a stable system, you should use the stable release, not a beta

donadigo, 2018-10-09:
@croozey it has been fixed with #97.

croozey, 2018-10-09:
That is **awesome!** My misunderstanding about testing the beta product. I've been dilligently applying all Juno updates daily, expecting that any fixes you guys make (such as this one) would flow through my beta version. I accept & agree wholeheartedly with your comments about not installing on production machines. I _am_ only using it on my test machines, which are all still exhibiting this bug, despite it being "closed" 29 days ago... A special thanks to Dan though, for taking the time to respond to me personally. That means a lot to me.

croozey, 2018-10-09:
Hmmm, I _always_ lock my screen when I leave my PC unattended. This bug _may_ still be present if the screen is manually locked (Windows-L), or no user logs in after a reboot. I deliberately left a couple of Juno test PCs unlocked and those PCs appear _not_ to have gone to sleep. I will do more testing...

Schyken, 2018-10-17:
I'm having this issue with Juno (post-release fresh install). I see above that the issue is marked as closed, but I don't think it should be. I can provide additional information about my system and install if necessary.

donadigo, 2018-10-17:
Reopening this for further investigation then.

croozey, 2018-10-17:
Just confirming, I purchased, downloaded & installed the lastest stable release on a test box. It still goes to sleep dispite being set to “never” suspend. I upgraded my other 2 Juno test boxes with apt, and they also still go to sleep. So yep, the problem persists...

cassidyjames, 2018-10-17:
@Schyken @croozey are you locking the screen before leaving it, or is the display turning off after a certain amount of time? What I suspect could be happening is actually https://github.com/elementary/greeter/issues/161

croozey, 2018-10-17:
As a sysadmin, I always lock my screen when unattended. I have just woken up one of my test box and will leave it logged in and get back to you if/when it goes to sleep. I have another test box which I will leave at the login prompt. Both are set to turn off the display after 10 minutes, but never suspend if inactive. I’ll get back to you soon.

Schyken, 2018-10-17:
> @Schyken @croozey are you locking the screen before leaving it, or is the display turning off after a certain amount of time? What I suspect could be happening is actually [elementary/greeter#161](https://github.com/elementary/greeter/issues/161)

I'm just letting it lock itself, not doing it manually. Though when I do lock it manually, the same thing does happen. It's kinda hard for me to tell which issue is which, sorry!

croozey, 2018-10-17:
The PC that was manually locked, stopped pinging after 20 minutes (2 x display sleep time).
The PC that was left at the desktop, locked itself after 10 minutes, then the display turned off after another 10 minutes, and finally stopped pinging after a further 10 minutes (30 minutes total).

ianmjones, 2018-10-18:
Original poster here ... can confirm that on a clean install of Juno "release" this issue still occurs for me. 😢 

Same steps as per original bug report, if I log in and leave my laptop alone it'll eventually turn off the screen and then some time later will suspend even though I have "Power" set to "Never" suspend on inactivity.

Mountainerd, 2018-10-23:
Are there any logs that can be provided that would assist?

Mountainerd, 2018-10-23:
@RodrigoOIiveira Huh. I just attempted the dconf-editor route and dconf shows that the **sleep-inactive-ac** setting was set to 0 (disabled) already. I had hope for a moment!

ianmjones, 2018-10-23:
The only way around this bug I've found so far is to set "Turn off display when inactive for:" to "Never".

Not ideal as it could wreck your display, but if you have something that needs to run for a while (e.g. syncing all the things from Google Drive after install like I was trying to do) and can switch between a few different apps now and then to make sure the screen gets some love or run something that keeps the display exercised, it is the only workaround I've found as even the various Caffeine style apps haven't worked for me.

RodrigoOIiveira, 2018-10-23:
> @RodrigoOIiveira Huh. I just attempted the dconf-editor route and dconf shows that the **sleep-inactive-ac** setting was set to 0 (disabled) already. I had hope for a moment!
You have to change "sleep-inactive-ac-type" to "nothing"

Mountainerd, 2018-10-23:
> You have to change "sleep-inactive-ac-type" to "nothing"
Yep, it's set, and still does its hibernate/sleep thing.

RodrigoOIiveira, 2018-10-25:
> > You have to change "sleep-inactive-ac-type" to "nothing"
> 
> Yep, it's set, and still does its hibernate/sleep thing.

Maybe because I used the command I found here https://elementaryos.stackexchange.com/questions/16519/elementary-os-ignoring-suspend-disable
But now, after the computer locks, the monitor won't suspend when the system is inactive.

Mountainerd, 2018-10-30:
Ah, I believe I tried that, too, but to no avail. This issue was starting to negatively impact workflow, so I had to move on to another distribution, but I'll keep watching this to see if it's resolved in the future.

croozey, 2018-10-30:
I’ve tried a **lot** of distros  in the last 23 years, but the truth is, I like ElementaryOS the best. I’d love to take advantage of some of the newer packages that come with 18.04 LTS, but unfortunately it looks like I’m going to be stuck on Loki for quite a while. This bug **has** to get fixed before I can switch... :-(

vaclavhlobil, 2018-10-31:
Agree with @croozey . I think this is CRITICAL bug. Imagine, your comupter is suspended in every 30 minutes when you go for a coffee break, when you talk with coleagues... and you can not do anything with it. With relation to other bug with nvidia (https://github.com/elementary/gala/issues/161), I can see glitches on my (lock) screen several times a day. You can imagine, how my impression of elementary OS looks like after few days...

croozey, 2018-11-08:
> This problem is because the power settings are used from the `gconf` configuration of the user `lightdm`, and not the current user.
> Just test:
> `sudo su`
> `su - -s /bin/bash lightdm`
> `gsettings get org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type`
> 
> If you change the settings of the user `lightdm`, the problem disappears

BINGO! —> “suspend”

croozey, 2018-11-08:
Error trying to set it to ‘nothing’:-

lightdm@eos001:~: gsetting set org.gnome.settings-deamon.plugins.power sleep-inactive-ac-type nothing

(process:13639): dconf-WARNING **: ... failed to commig changes to dconf: Error spawning command line “dbus-launch ...”: Child process exited with code 1

Close, but no cigar... :)

unameisfine, 2018-11-08:
> Error trying to set it to ‘nothing’:-
> 
> lightdm@eos001:~$ gsettings set org.gnome.settings-deamon.plugins.power sleep-inactive-ac-type nothing
> 
> (process:13639): dconf-WARNING **: ... failed to commig changes to dconf: Error spawning command line “dbus-launch ...”: Child process exited with code 1
> 
> Close, but no cigar... :)

use:
`dbus-launch gsettings set org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type nothing`

3mpireGamer, 2018-11-10:
I have the same issue. Trying to run a Server and since it keeps suspending the Server shuts down. Both on battery and plugged in are set to Never.

croozey, 2018-11-14:
> use:
> dbus-launch gsettings set org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type nothing
> 
No protocol specified

2whlrider, 2018-12-08:
I'm confirming that this is still a problem in Juno. My desktop ignores any time settings I load into the spring fields and suspends after 15 minutes. This wasn't a problem in Loki.

wramby, 2018-12-25:
> use: dbus-launch gsettings set org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type nothing
> Response from running above command:- No protocol specified

I was getting this too. All I did to fix it was open a seond terminal and run the same command with sudo, i.e.:

`sudo dbus-launch gsettings set org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type nothing`

vaclavhlobil, 2018-12-28:
> I was getting this too. All I did to fix it was open a seond terminal and run the same command with sudo, i.e.:
>
> `sudo dbus-launch gsettings set org.gnome.settings-daemon.plugins.power [...]

This doesn't work for me at all. elementary OS still suspends.

Schyken, 2019-01-14:
@benjaminjack I don't think there's any solution yet, but the one thing I have come to find is that if I have something I need to run and I need the machine to keep from suspending, I can use `systemd-inhibit` either with the command I'm running or on its own until I kill it. It's an inelegant way to try and smooth over that rough patch, but it's all I've been able to do so far. This is definitely a critical bug, but I have absolutely no idea what causes it or what it might take to fix it.

`systemd-inhibit --mode=block --what=sleep <command>` is what I've had to do. If you want, you can replace the command with `sleep infinity` so it just blocks system suspend until you kill `sleep` yourself.

cassidyjames, 2019-01-14:
@vaclavhlobil different people work on different things.

cassidyjames, 2019-01-14:
@ianmjones if you are in a secure environment, you can also turn off locking under [Security & Privacy settings](settings://security/screensaver). This way your display can go to sleep but the machine won't lock unless explicitly locked (with <kbd>⌘</kbd> <kbd>L</kbd> or from the panel)

ianmjones, 2019-01-14:
@cassidyjames That's a great idea, I can definitely give that a whirl, thanks for the tip.

unameisfine, 2019-01-14:
What is your result after executing these commands?
```
sudo su
su - -s /bin/bash lightdm
gsettings get org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type
```

> Yikes, I just encountered this and it's is a serious bug. Any timeline on a fix? I do deep learning work and my machine often needs to run for days to train a model. I can't have it going to sleep on me.

Mountainerd, 2019-01-15:
@cassidyjames The issue isn't system locking, the issue is system sleep. If it locked, it wouldn't be an issue. The issue is we leave workloads going and eOS sleeps after 30 minutes or so.

Schyken, 2019-01-15:
@Mountainerd I think the point of this is that the sleeping only happens once the session is locked, not before. So if it never locks, sleep won't come after.

croozey, 2019-01-15:
The “gsettings” work-around under the lightdm user account does work, but I really would like to see this annoying bug get fixed - it’s been ages! It should be easy to add a tick box to “Set as system default” in the power settings screen, which would make this change under the lightdm user for you. I’m sure this would only be a 5 minute job for someone like Daniel or Cassidy :)

albergh, 2019-01-21:
> > > Yikes, I just encountered this and it's is a serious bug. Any timeline on a fix? I do deep learning work and my machine often needs to run for days to train a model. I can't have it going to sleep on me.
> > 
> > 
> > What is your result after executing these commands?
> > ```
> > sudo su
> > su - -s /bin/bash lightdm
> > gsettings get org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type
> > ```
> 
> I can confirm that this worked for me. But it would be nice if the toggle switch in settings actually worked.

How did you make it work? I type those comands and I only get as result "suspend". I've tried to modify through gsettings but it seems I can't modify anything related to the lightdm user since I don't have the password

croozey, 2019-02-06:
> > > > Yikes, I just encountered this and it's is a serious bug. Any timeline on a fix? I do deep learning work and my machine often needs to run for days to train a model. I can't have it going to sleep on me.
> > > 
> > > 
> > > What is your result after executing these commands?
> > > ```
> > > sudo su
> > > su - -s /bin/bash lightdm
> > > gsettings get org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type
> > > ```
> > 
> > 
> > I can confirm that this worked for me. But it would be nice if the toggle switch in settings actually worked.
> 
> How did you make it work? I type those comands and I only get as result "suspend". I've tried to modify through gsettings but it seems I can't modify anything related to the lightdm user since I don't have the password
> 
> Edit: Working with a user's help from the telegram's group. If someone's still have the problem just do exactly as this (tested with two computers, one laptop and one desktop):
> 
> > sudo su
> 
> enter, then your password
> 
> > su - -s /bin/bash lightdm
> 
> enter
> 
> > dbus-launch gsettings set org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type nothing
> 
> enter
> 
> > gsettings get org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type
> 
> enter
> 
> it should return _'nothing'_

I just got email notification of your update/edit - Sorry, I missed your original question. My bad - I clearly rushed the cut/paste and missed the fact that my "gsettings" command was a "get" request. Sounds like others have come to your rescue - thanks! :) Changing "get" to "set" and adding "nothing" to the end should do the trick. The command has to be executed under the "lightdm" user, hence all the sudo/su stuff.

croozey, 2019-02-06:
So, a comment from the code maintainers about this bug would be really great :) It just would be nice to understand why there isn't just a quick fix for it...

naftulikay, 2019-02-06:
I am affected by this as well. I'm on a Thinkpad 460S and all hardware works out of the box, thankfully, have no problems whatsoever going to suspend and back.

Here are my `gsettings`:

```
org.gnome.settings-daemon.plugins.power sleep-inactive-ac-timeout 0
org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type 'nothing'
org.gnome.settings-daemon.plugins.power sleep-inactive-battery-timeout 0
org.gnome.settings-daemon.plugins.power sleep-inactive-battery-type 'nothing'
```

I'm noticing that while plugged in, my device suspends rather quickly, even though I have sleep disabled for both AC and battery power.

Thinkpads usually have two batteries:

```
/sys/class/power_supply/BAT0
/sys/class/power_supply/BAT1
```

Both batteries are fully charged. 

Nerds: is there a syscall I can use to prevent suspend? I could definitely write a little application in Rust or C to use systemd conditions to enable/disable suspend on AC presence.

carniz, 2019-02-07:
> It should be easy to add a tick box to “Set as system default” in the power settings screen, which would make this change under the lightdm user for you.

This is the best suggestion I've seen when it comes to a permanent fix.

carniz, 2019-02-07:
I can confirm that this issue also happens on my machine (a Dell Precision 5510) after upgrading to Juno. What's making it worse is that I can't login after waking it up from suspension and have to reboot to get back to normal. I have configured the machine to require a Yubikey present for all authentication (see https://github.com/carniz/ubuntu-yubikey-setup) but I don't know if that has anything to do with it since I never tried logging in after suspension before enabling Yubikey support.

tuliopa, 2019-02-07:
I had the same issue, my machine is a HP ProBook-450-G3 

this command worked for me
`sudo gsettings set org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type nothing`

Thanks for the command.

thomasvjohansen, 2019-03-19:
still a problem

croozey, 2019-03-24:
Closing this ticket without a fix or even an explanation speaks volumes about the respect, support & commitment shown to the user community. The complete apathy shown on this issue is the reason why I have stopped evangelizing Elementary OS 5 to family, friends & co-workers. I'm now telling them to check out MX Linux instead...

Mountainerd, 2019-03-24:
@croozey This issue wasn't closed. A different issue that references this one was closed. And the other issue was closed for being a duplicate of this one, so, really, they're consolidating the tracking of the work around this (still) open issue.

vaclavhlobil, 2019-03-25:
Although this bug is not closed, I have exactly the same feelings as @croozey. From enthusiasm, effort, through upset, anger, to giving up and finaly resignation. I hadn't reported few another bugs, because of "Why? Nobody cares." Core of elementary OS team is playing around with icons, mashing up anonymous browser, which nobody needs, tweeting how eOS is the best etc. Meanwhile users have self-hibernating computers for a months. We were waiting years for this "when it's ready" release. And when we realized, it's not ready yet, we should probably wait next year for fixing critical bug. And in every response of eOS team you can feel, it's only a toy for a few boys. Responses like "different people work on different things", instead of "wait! it's bad! give us a few days, we have to fix it asap!". Phew.

croozey, 2019-03-25:
@vaclavhlobil Hear, hear...

thomasvjohansen, 2019-03-25:
just found this, I will try it tonight...to see if its a sort of workaround.

https://github.com/elementary/greeter/issues/161#issuecomment-426061600

marianitn, 2019-03-25:
I can't agree more with @vaclavhlobil. Although this might be tagged "offtopic" soon, I have a huge disappointment with this project. I'm an eOS user since Jupiter, and have reported or commented since several bugs, most of them being just ignored or kicked for "next release". Some of them [1] with more than 5 years are STILL PRESENT. And some others like this one, combined with some strange ligth-locker bug that doesn't unlock the session after suspend, turns the OS UNUSABLE: it suspends whent it wants, it doesn't come back after resume. You loose your work, and have to reset the session or REBOOT, like on the good old windows days.
And that "I don't care" attitude from developers totally discourages people from participating or contributing to the project. So, no more funding from me, no more bounties, and no more bug reporting.
I think you really need to reconsider your focus: new shiny features are nice and fun, but **support matters**. I'm not talking about "rough edges" bugs, but **critical** bugs that should be treated as... well, critical!

[1] https://bugs.launchpad.net/elementaryos/+bug/1220179

croozey, 2019-03-26:
> just found this, I will try it tonight...to see if its a sort of workaround.
> 
> [elementary/greeter#161 (comment)](https://github.com/elementary/greeter/issues/161#issuecomment-426061600)

TLDNR? For everyone stills truggling with this issue, the below **does work**. Its just a shame the developers can't seem to find the time to fix the power settings app (taken from @naftulikay above):-

* become the superuser

`sudo su`

* launch a bash shell as lightdm

`su - -s /bin/bash lightdm`

* use dbus-launch to launch the *gsettings* command. within the active dbus session the gsettings command sets the given key to the special value 'nothing', which is the setting which should disable suspend when connected to a power source:-

`dbus-launch gsettings \`
`    set org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type nothing`

This fix has worked on all machines I've tried it on so far.

cassidyjames, 2019-03-27:
>It should be easy to add a tick box to “Set as system default” in the power settings screen, which would make this change under the lightdm user for you.

I believe this is similar to the Language & Region settings. Would that be a satisfactory fix, @danrabbit? @codygarver I'm not sure how the DPMS stuff (https://github.com/elementary/dpms-helper/pull/9) works, but I see it was closed without being merged. Do you know what's up there?

thomasvjohansen, 2019-03-27:
Croozey, didn't work for me😣

croozey, 2019-03-27:
@thomasvjohansen, I will do a fresh install & apply all updates on a spare physical PC for you today to confirm this fix still works. All the PCs I've run this fix on so far don't go to sleep anymore when the display locks. My understanding is that the login/lock screen runs under the lightdm user. Therefore, when the screen locks or if no-one logs in after a reboot, it is the preferences of the lightdm user that matter. Your personal preferences only control if/when the screen will lock after you log in.  Let me confirm and get back to you...

[4 more replies]

PR #97 (2018-09-10): Also set sleep-inactive action type enums
I think this fixes #89 from my limited testing.

It would seem that despite the GSD schema saying that 0 means never, you also have to set the "do nothing" option on the relevant key too.
PR #123 (2019-09-20): Set AccountsService power keys
Fixes #89 when used in conjunction with https://github.com/elementary/greeter/pull/343

Requires https://github.com/elementary/default-settings/pull/134 to work but won't crash without it.
closed 2019-09-20 by danrabbit