elementary/settings-power #101: Enable hibernation using power-button does not work

RussH, 2018-10-22:
I have Hibernation working successfully, I can invoke it using "systemctl hibernate" (you do need a little work to get it operational). However it's not invoked by pressing the power button. Is there some config file to amend someplace?

Note: I use hibernation rather than suspend as the battery is fairly dead and doesn't last a few days with suspend.

igalic, 2018-11-10:
I would also like to use hibernation rather than suspend because i have my disk encrypted. suspend doesn't protect it from anything.

cassidyjames, 2019-05-03:
This is either a duplicate of #78, or should be moved to the OS repo since this plug is doing what it's supposed to regarding hibernation, it's just not working in the OS itself.

closed 2019-05-03 by cassidyjames