elementary/settings-power #104: Brightness adjustment

macgarcia, 2018-11-18:
Brightness adjustment, through F11 and F12, interface stops.

peteruithoven, 2019-04-30:
I'm afraid this has nothing to do with the power settings.

closed 2019-04-30 by peteruithoven

FernandoLGL, 2019-06-12:
What does this have to do with, then? I can't find an appropriate section. I was about to report the same problem. I have elementary on 2 PCs, and in both of them, adjusting brightness through keyboard shortcuts causes elementary to basically crash(actually right now it doesn't crash anymore, in the beginning it did, now it's getting better but by better I just mean that I don't have to force shutdown my computer anymore. It still "crashes" the system for like a full minute)

peteruithoven, 2019-06-12:
If I had to guess this probably happens in Gala (the project that shows notifications) or Mutter. 
It kind of looks like the following issue, except that you're experiencing a much longer freeze. https://github.com/elementary/gala/issues/369

To clarify with "crashes"  you mean freezes your system momentarily (like a full minute)? 

Seeing we're not hearing about this issue from a lot of people it would be very useful to help us figure out why this happens. Does it for example happen on a clean system (you could try starting a trial session from a usb stick. 
Does it only happen when you alter brightness or also with other keyboard shortcuts? 
Does it also happen when you alter the brightness though the power indicator or the power settings?