elementary/settings-power #105: Changing brightness does not work sometimes

janxkoci, 2019-01-25:
After some recent update (I noticed this about a week ago) my brightness controls sometimes do nothing after booting the system. I tried keybord shortcuts, indicator and even switchboard, none of the methods changes the brightness. If I then shut down the laptop and boot it up again, it usually starts working. This issue was not present few weeks ago for sure.

The UI controls are interactive, I get notifications with shortcuts, the sliders in indicator and switchboard do slide. It's just the screen itself doesn't change its brightness.

Is there any way I can debug this further?

OS: Juno
HW: Acer Aspire One Cloudbook 11, Acer ComfyView LCD

stem2580, 2021-08-11:
i also have this problem in odin

janxkoci, 2021-08-17:
I gave that netbook to my dad (it still has Juno, as I was unable to reinstall anything on it for some weird reason), so I cannot test it at the moment. I don't have this issue on my other laptops.