elementary/settings-power #106: Upgrading from 2.3.4 to 2.3.5; unable to use dock

asiral, 2019-01-29:
I have just upgraded from 2.3.4 to 2.3.5.  I am now unable to use my dock with my laptop.  Is there anyway to rollback?  When I put my laptop on my dock now, neither monitor works and neither does the laptop monitor.  Taking it off the dock, the laptop monitor stays black (no backlight).  I have to force a shutdown with the power button.  Booting with the laptop in the dock displays the same behavior.

peteruithoven, 2019-01-29:
Thanks for reporting, but how does your laptop monitor's behavior relate to the Power system settings? Are you sure that was the only thing that was updated? 
On installing specific versions and also holding it there: https://askubuntu.com/a/138327/519324
Trying this might also clarify whether this is caused by the Power system settings.

asiral, 2019-01-29:
The power settings control what happens with the monitor when the lid is closed.  If I keep my lid open on the dock, it is fine.  Not until I close my lid does nothing work.  Here is from my history log.  After the update this morning, everything was fine.  It was not until the second update today, during which only switchboard-plug-power was updated that I experienced this behavior.
`Start-Date: 2019-01-29  10:09:07
Commandline: packagekit role='update-packages'
Requested-By: michael (1000)
Install: linux-image-4.15.0-44-generic:amd64 (4.15.0-44.47, automatic), linux-headers-4.15.0-44:amd64 (4.15.0-44.47, automatic), linux-headers-4.15.0-44-generic:amd64 (4.15.0-44.47, automatic), linux-modules-extra-4.15.0-44-generic:amd64 (4.15.0-44.47, automatic), linux-modules-4.15.0-44-generic:amd64 (4.15.0-44.47, automatic)
Upgrade: linux-headers-generic:amd64 (4.15.0.43.45, 4.15.0.44.46), linux-libc-dev:amd64 (4.15.0-43.46, 4.15.0-44.47), linux-image-generic:amd64 (4.15.0.43.45, 4.15.0.44.46), libasound2-data:amd64 (1.1.3-5ubuntu0.1, 1.1.3-5ubuntu0.2), switchboard-plug-applications:amd64 (2.1.4+r304~ubuntu5.0.1, 2.1.5+r342~ubuntu5.0.1), gala:amd64 (0.3.2~r679+pkg45~ubuntu5.0.1, 0.3.2~r680+pkg45~ubuntu5.0.1), pantheon-photos:amd64 (2.6.1+r3815+pkg50~ubuntu5.0.1, 2.6.2+r3858+pkg51~ubuntu5.0.1), libgala0:amd64 (0.3.2~r679+pkg45~ubuntu5.0.1, 0.3.2~r680+pkg45~ubuntu5.0.1), libasound2:amd64 (1.1.3-5ubuntu0.1, 1.1.3-5ubuntu0.2), virtualbox-5.2:amd64 (5.2.24-128163~Ubuntu~bionic, 5.2.26-128414~Ubuntu~bionic), linux-generic:amd64 (4.15.0.43.45, 4.15.0.44.46)
End-Date: 2019-01-29  10:10:22

Start-Date: 2019-01-29  13:33:39
Commandline: packagekit role='update-packages'
Requested-By: michael (1000)
Upgrade: switchboard-plug-power:amd64 (2.3.4+r609~ubuntu5.0.1, 2.3.5+r634~ubuntu5.0.1)
End-Date: 2019-01-29  13:33:40`

I tried to follow that article before posting, but specifying 2.3.4 does not work.  Has it been removed from the elementary repository?  I have tried with ":adm64", without, also with the full version after the "+", without, and multiple permutations but no luck.  Can you advise the exact command I should use to downgrade to 2.3.4?  Thanks for the quick followup.

asiral, 2019-01-29:
Additionally, with the lid is open, the laptop monitor and two external monitors will display fine.  It is not until I close the lid when on the dock that the issue occurs.

peteruithoven, 2019-01-29:
Valid point. 
The thing is though, there where only translation updates: https://github.com/elementary/switchboard-plug-power/compare/2.3.4...2.3.5

`apt-cache showpkg switchboard-plug-power` shows `2.3.4+r609~ubuntu5.0.1` as the previous version, but I also can't install that version, `apt-get` says it can't be downloaded:
```
$ sudo apt-get install --reinstall switchboard-plug-power=2.3.4+r609~ubuntu5.0.1
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Reinstallation of switchboard-plug-power is not possible, it cannot be downloaded.
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
```

peteruithoven, 2019-01-29:
Looks like someone found a solution to install an older version: https://elementaryos.stackexchange.com/a/18128/3892

Please update this issue on whether that solves your issue.

asiral, 2019-01-30:
I compared the versions first and I knew that made it unlikely to be the issue.  But it seemed weird that any update from the morning would break so much functionality (as I said, I cannot even shut down the computer when on the dock and using all three screens).

I posted that stack exchange question.  I am glad someone got back to me.  I'll resolve that question as I was able to downgrade and it did not resolve the issue (not too surprising).  Hopefully it will be useful for others at least looking to downgrade an elementary os package.

Going through the (longer) list of packages installed from that morning, it seems to have been a kernel issue.  Booting into the previous kernel, the issue is resolved.

closed 2019-01-30 by asiral