elementary/settings-power #110: Brightness controls - missing

johans3, 2019-04-30:
I have an issue with **brightness controls**, they are **completely missing**. 
I already posted my issue on **stack-exchange** and got very nice support. But we were not able to solve the issue

[https://elementaryos.stackexchange.com/questions/18860/no-brightness-controls-after-fresh-install/18862?noredirect=1#comment21738_18862
](url)

_My Computer_: Mac OsX, 
_Elementary_: Juno, i installed the newest updates, 
_Plugins_: purged some plugins on stack-exchange

_Current context_ (With Tweaks installed  ^^ ) 
[image]

I tried to install 3rd party plugins to add brightness settings by my own. These applications were outdated and did not work anymore. The only application available in the AppCenter does not dim enough, not even on max, and interferes with NightLight. If dimmer is on, NightLight switches off and vice-versa

IamLizu, 2019-04-30:
I think the correct link to the Stack Exchange post is - https://elementaryos.stackexchange.com/questions/18860/no-brightness-controls-after-fresh-install

peteruithoven, 2019-04-30:
Not sure why you're showing the Nightlight indicator. That's something else. 
The power indicator menu with brightness control:
[image]

I'm assuming it won't show up for you there seeing it's not visible in the System Settings. 

Mac OsX is a operating system. It tells us 0 about your hardware.

johans3, 2019-04-30:
The nightlight wingpanel is also a kind of screen luminosity control, so i assumed Brightness controls share the same panel.
I used Juno for 2 days out of the box, but after looking thouroughly for brightness controls which i could not find, i tried to install some 3rd pary app to get some brightness control.
The beauty of Juno is its clean uncluttered interface. There is nothing confusing. 

Secondly i thought…Maybe the battery panel is surpressed because my computer is not a laptop.

[image]

peteruithoven, 2019-04-30:
A relevant Stack Exchange question: https://elementaryos.stackexchange.com/questions/5465/how-to-reduce-the-brightness-of-a-desktop-pc

Looks like the power settings look for a file in `/sys/class/backlight/`, see: 
https://github.com/elementary/switchboard-plug-power/blob/master/src/MainView.vala#L257-L277
Do you have something in `/sys/class/backlight/`?
It's not dependent, as far as I understand it on whether the computer is a laptop.

johans3, 2019-04-30:
There is nothing inside  `/sys/class/backlight/`
ok, i will create a the file there, logout and report back

peteruithoven, 2019-04-30:
That's not how that works. You can't just add a file and have it work. 

Please read for example: https://askubuntu.com/questions/894465/changing-the-screen-brightness-of-the-external-screen

I guess the question would be whether we can integrate Brightness Controller into the power settings, but then I wonder why even Ubuntu didn't do that.

johans3, 2019-04-30:
indeed it does not work this way. I created the file in Downloads and tried to move it to dest directory. Access denied even with sudo
```
cd Downloads
sudo cp MainView.vala /sys/class/backlight/MainView.vala
```

johans3, 2019-04-30:
i read somewhere online that Ubuntu derivatives suffer from common problems, like the one controlling brightness  adjustments. My battery preference is surpressed somehow, but this issue were there from day one on. Following the installer process, i noted that Juno had ignored various packages during its first install. 

I configured nothing special in the Juno installer on my USB stick (then installing over my HD partition), except the reformatting of my Fat partition i had created before on my Mac, to a linux partition - ext4 to be exact.

johans3, 2019-04-30:
should i see something useful in: `systemctl list-units --type service` ?
the only line with error is:
`xpra.service                                          loaded failed failed  Xpra System Server`
could not find anything related by searching for keywords like: battery, brightness or their like :(

PR #251 (2024-05-14): Mainview: simpler backlight detection
If there's no backlight then screen brightness will be -1

Maybe fixes #110
closed 2024-05-14 by danirabbit