elementary/settings-power #111: Add interactive "Promot to Shut Down" option for Power Button

cassidyjames, 2019-05-03:
This is the behavior common on desktops where pressing the power button shows a GUI prompt to shut down, exposed as "interactive" in GSD. Could potentially supersede #21.

lohxx, 2019-05-16:
Can I work on this issue?

cassidyjames, 2019-05-16:
@lohxx absolutely!

lohxx, 2019-05-17:
How can i procede with code?

cassidyjames, 2019-05-17:
@lohxx you can check out the README at https://github.com/elementary/switchboard-plug-power. It does assume you're on elementary OS. You should also follow the directions at "The Basic Setup" section of the developer guide: https://elementary.io/docs/code/getting-started#the-basic-setup

Once you're all set, you can fork this repository, make the change, and then submit a Pull Request!

lohxx, 2019-05-17:
Oh okay, thank you!

PR #115 (2019-06-08): Added interactive "Prompt to shutdown" option in power button menu
Closes: #111  

[image]
closed 2019-06-08 by danrabbit