elementary/settings-power #117: Lid closed action customizable in both Plugged in and On Battery states

mualhadi, 2019-06-07:
Migrating from Win10 to Elementary was great and the only downside is power management. Tere's no differentiation on options for (battery vs AC)

On Win10 i used to set my laptop to never sleep when i close lid on AC. And to sleep when lid close on battery. This work flow is essential in an open workplace where you need to perform lengthy tasks or uploads while away.

In Elementary there's option for lid close action in general but without AC/Battery customization. We need better power management in eOS

peteruithoven, 2019-06-07:
Relevant code:
https://github.com/elementary/switchboard-plug-power/blob/master/src/Widgets/LidCloseActionComboBox.vala
https://github.com/elementary/switchboard-plug-power/blob/master/src/LogindHelper.vala

danirabbit, 2024-02-13:
I'm not sure why we're using the logindhelper here for these settings actually. It looks like GNOME Settings daemon covers settings for line power, battery, and an option here for external displays

[image]

danirabbit, 2024-02-13:
Okay, so I tried to implement the UI for this according to these settings, but they don't actually exist upstream: https://gitlab.gnome.org/GNOME/gnome-settings-daemon/-/blob/master/data/org.gnome.settings-daemon.plugins.power.gschema.xml.in?ref_type=heads

So it's not doing anything in my branch :sweat_smile: Guess that's why we don't use those keys