elementary/settings-power #122: Computer always goes to sleep

kakogoxepa, 2019-08-25:
I set the settings everywhere so that the computer does not fall asleep. But he continues to fall asleep after about half an hour. I put the torrent to download and left. When i back the computer slept. Always like this. Correct it urgently, so it’s not possible to work normally. Since when the computer comes out of sleep, ripples and glitches appear on the screens, i have to reboot to make everything normal.
[image]

kgrubb, 2019-08-25:
This is possibly a duplicate of #89, can you try the steps in this comment: https://github.com/elementary/switchboard-plug-power/issues/89#issuecomment-461169400 and see if that resolves your issue?

naggie, 2019-12-27:
I also had this issue with my desktop computer. The workaround described works but it not scriptable due to the usage of `su` requiring subshells. I was able to use sudo instead after specifying the lightdm home directory with:

```
sudo -u lightdm env HOME=/var/lib/lightdm dbus-launch gsettings set org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type nothing
```

naggie, 2019-12-29:
Further to this, if I lock my PC it says awake as expected; but after I log in the following morning it goes into standby

naggie, 2020-01-05:
I ended up disabling the ability to suspend altogether using systemd, as explained here: https://elementaryos.stackexchange.com/questions/16519/elementary-os-ignoring-suspend-disable

`sudo systemctl mask sleep.target suspend.target hibernate.target hybrid-sleep.target`

Not suitable for a laptop of course, but I'm running on a desktop do it's OK for me at least.

kgrubb, 2020-04-15:
This should be resolved now that the changes in https://github.com/elementary/switchboard-plug-power/pull/123 have been merged in and released!

closed 2020-04-15 by kgrubb

naggie, 2020-04-15:
Great to hear that, thanks @kgrubb -- I can use Elementary again!