elementary/settings-power #131: Close lid to encrypt | Hard disk encryption in Linux suspend mode

4jNsY6fCVqZv, 2019-12-18:
## Prerequisites
- [x] I have searched open and closed issues for duplicates.

## Feature
**Is your feature request related to a problem? Please describe.**
"hard disk encryption only protects user's data when their machine is shut down"

**Describe the solution you'd like**
"When you close the lid of your notebook, it goes into sleep/suspend mode. All processes are frozen and don't need to access your hard disk anymore. We use this opportunity to clean the keys of your encrypted devices and suspend them as well. Therefore, the data on your hard drive is protected. When resuming your computer, you must re-enter the password of your encrypted volumes. But then you're just where you've been working before."

I'm not sure if the Switchboard Power Plug is the right place for this feature request.
It's also possible to combine it with a new Encryption Pane.
I'm just thinking of a simple on/off switch in the Power Plug settings.
Of course it should also be possible to unlock with a Nitrokey if the "Close lid to encrypt" option is activated.

**Existing work**
please take a look a https://fosdem.org/2020/schedule/event/dip_close_lid_encrypt/

cassidyjames, 2019-12-18:
I _think_ this is lower-level than we can realistically work on ourselves. If this comes to Debian (and Ubuntu specifically), then it would be much more likely that we could enable it.

4jNsY6fCVqZv, 2019-12-18:
Thanks for your feedback!

I share your thoughts about an implementation.
Since the project is funded by the prototype fund, it looks good that it can be realized.

For now the project page is unfortunately in German, but the linked presentation at FOSDEM 2020 will probably be in English.

https://prototypefund.de/project/close-lid-to-encrypt/

I quote: "The developed solution is to be integrated into Debian and made configurable via the Gnome system settings. [...] Since the solution is implemented in Debian, it will reach all users of Debian derivatives (such as Ubuntu, Tails, Linux Mint) in the medium term, as soon as they upgrade to a current version of their operating system."

Here is a link to their GitLab project repository:
https://salsa.debian.org/mejo/cryptsetup-suspend

4jNsY6fCVqZv, 2022-03-07:
To keep this issue up to date: `cryptsetup-suspend` has arrived in Jammy.
https://packages.ubuntu.com/jammy/cryptsetup-suspend

According to the developer, the package is definitely ready to be deployed on systems. He recommends sending feedback in form of bug reports directly to the maintainers of the Ubuntu packages.

The package is not active by default. With the manual installation of the package, the function is automatically enabled.