elementary/settings-power #137: Notebook suspendendo mesmo quando configurado para não suspender

andrelopesrigo, 2020-02-16:
O notebook está suspendendo mesmo estando configurado para nunca suspender quando ligado na tomada. Estou usando o Elementary 5.1.2.

Kisuke-CZE, 2020-03-04:
Yeah, some update broke this.

It's again the old situation, when user lightdm has different settings than user logged in.
Unfortunately, even if you fix it manually (which is not hard), on next reboot settings are restored.
Any idea what component is restoring those settings?

Temporary fix (until next reboot) is:
```
su - -s /bin/bash lightdm
dbus-launch gsettings set org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type nothing
dbus-launch gsettings set org.gnome.settings-daemon.plugins.power sleep-inactive-ac-timeout 0
```

After reboot settings are back to
```
sleep-inactive-ac-type = suspend
sleep-inactive-ac-timeout = 1200
```

lenemter, 2024-02-10:
Closing because I cannot reproduce the issue on my OS 7.1.

closed 2024-02-10 by lenemter