elementary/settings-power #139: Hera: "suspend when inactive" doesn't work

wdysun, 2020-03-06:
after last `dist-upgrade` to eOS 5.1.2 Hera, the issue with the system suspending after some minutes of inactivities keep up again. I had solved it in the past (but I can't remember how), but after last update it is back:

on my desktop I set    Sysem Settings > Power >  Suspend when inactive for = "Never"

however it keeps suspending.

## Prerequisites
- [ ] I have searched open and closed issues for duplicates.

## Describe the bug
A clear and concise description of what the bug is.

If applicable, add screenshots or screen recordings to help explain your problem.

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

### Screenshots or screen recordings

### Logs

## Platform Information

- [ ] I'm using the latest version from git that I've manually compiled
- [ ] I'm using the latest released stable version

## Additional context

vaclavhlobil, 2020-03-06:
Neverending story... Same problem here on brand new installation of elementary OS 5.1.

landintrees, 2020-03-06:
I think this is the greeter not respecting the power settings, there's a workaround in this issue:
[](https://github.com/elementary/greeter/issues/161#issuecomment-426061600)

wdysun, 2020-03-06:
I am not sure why but the following worked for me

```bash
sudo dbus-launch gsettings set org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type nothing
```
let's hope this issue doesn't come back, but just in case I noted the solution above.

Kisuke-CZE, 2020-03-09:
Hi,

WA works but only for limited time. Something is still changing `sleep-inactive-ac-type` for lightdm user back to `suspend`.
I did workarround with gsettings. But when I came back to my work laptop today, it was suspended again.

Maybe I will have to mask suspend (maybe also sleep) target in systemd as was suggested in other threads. But I would preffer if I do not have to. Because this was I will disable suspend/sleep completely (even if I want to sleep laptop intentionaly).

gsupp, 2020-03-19:
Kisuke-CZE, I'm seeing the same thing. I've used these commands in the past to prevent suspending at the greeter:
`sudo su`
`su - -s /bin/bash lightdm`
`dbus-launch gsettings set org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type nothing`

And verifying the setting with: `gsettings get org.gnome.settings-daemon.plugins.power sleep-inactive-ac-type` but it keeps getting changed back to `'suspend'` on Hera.

biemster, 2020-04-09:
I ran into the same issue again as well, after some update brought it back last week. I sunk a lot of time in it two years ago when I first installed Juno, and wrote about my solution, thinking only very few people were affected considering the response on [my post here](https://elementaryos.stackexchange.com/a/16576/15966).

Seeing lots of people complaining now, my workaround now is to ditch the pantheon greeter all together, I went with the lightdm-gtk-greeter:
`sudo apt install lightdm-gtk-greeter`
Had no issues so far. (i did not like the new pantheon greeter that much anyway, although it still looks better than the lightdm greeter.)

lenemter, 2024-02-10:
Closing because I cannot reproduce the issue on my OS 7.1.

closed 2024-02-10 by lenemter

chinarut, 2024-04-03:
I can reproduce this issue on eOS 7.1 on an early 2015 MacBook Pro.

I've tried changing settings as the lightdm user but they don't stick. My workaround is to disable suspend entirely (which is not ideal but fine for now):

```
systemctl mask sleep.target suspend.target hibernate.target hybrid-sleep.target
```

is there any specific information I can provide to get to the bottom of this?