elementary/settings-power #141: sleepping mode

cwbyun, 2020-04-09:
I used elementary os a few days ago.

I don't want to use sleeping mode.

But the OS goes to the sleep mode regardless of the configuration.

Let me know how to prevent going to sleeping mode.

Following is output after commend "uname -a"
Linux 5.3.0-45-generic #37~18.04.1-Ubuntu SMP Fri Mar 27 15:58:10 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

jeremypw, 2020-04-09:
Have you tried Settings/Power/Suspend when inactive for = Never? for Plugged and and/or Battery?

Yofiel, 2020-04-30:
> Have you tried Settings/Power/Suspend when inactive for = Never for Plugged and and/or Battery?

This may be a workaround, but it is not a fix. This problem is happening frequently with multiple users.

ppearson, 2021-02-27:
I've just installed 5.1.7 on a Dell XPS 13 to test Elementary OS (haven't tried it in a few years), and am having this issue: regardless of the status of the "Suspend when inactive for" combo boxes for both plugged in and on battery (I've set them both to "Never") in the Power section of System Settings, it still suspends.
Even after a reboot after changing the settings.

lenemter, 2024-02-10:
Closing because I cannot reproduce the issue on my OS 7.1.

closed 2024-02-10 by lenemter