elementary/settings-power #145: blank screen upon resume from suspend

bnutt123, 2020-04-30:
Using a laptop plugged into an external monitor by display port.  When computer resumes from suspend only a blank screen is available.  Unable to get to any type of graphical environment.  Force restart only way to get back to graphical environment.

msdisme, 2020-08-09:
Same problme on Imac - Advanced Micro Devices, Inc. [AMD/ATI] Broadway PRO [Mobility Radeon HD 5850] -  Quad-Core Intel® Core™ i7 CPU         870  @ 2.93GHz --   elementary OS 5.1.7 Hera

im-barney, 2021-05-01:
I have this same problem on the built in monitor on a Dell Latitude 7490. 
Intel UHD 620 Graphics. 
Elementary OS 6 EA version.

lug-gh, 2021-12-16:
Same for me, it doesn't matter if the monitor is connected directly to the HDMI port or with a docking station (HDMI/DP).

I was able to switch to another tty with CTRL+ALT+F1 (F7 or F8 is the elementaryOS GUI)
So in theory I could restart a service or sth like that, but I don't know which one.

[image]
[image]