elementary/settings-power #149: Pantheon system panel tray

AndrzejL-eu, 2020-05-10:
Hello,

## Prerequisites
- [ ] I have searched open and closed issues for duplicates.

I did my research and got it to work but this could be done easier by just adding it by default.

## Feature
**Is your feature request related to a problem? Please describe.**

So. I like to use a system tray in the top panel.

I use it for some of the apps that support it like wire or smplayer.

[image]

**Describe the solution you'd like**

The solutions that I've found kinda worked however they left me with icons in the tray that were different sizes and with huge gaps between them.

A solution that I've found working well is described here

https://github.com/mdh34/elementary-indicators/issues/1#issuecomment-546731215T

The icons are roughly the same size and they are spaced nicely.

Could we please have this implemented in pantheon by default? Maybe a option in the settings that will enable it (for those that dislike system tray).

**Existing work**

**Describe alternatives you've considered**

**Additional context**

Thanks in advance

Kind regards.

Andrzej

cassidyjames, 2020-05-10:
A system tray is a legacy API that was never guaranteed to exist in FreeDesktop environments or elementary OS. It has significant issues with accessibility, stability, and future-proofness. Instead, desktops (including elementary OS) have been moving to more specific, well-defined APIs that don't define their design as part of the API, but define their intent so that desktops are more free to implement them in smart and accessible ways. You can read about some of those APIs available to apps on elementary OS in this developer blog post: https://blog.elementary.io/developer-tips-backgrounding-system-integration/

You can also explore some of these APIs here: https://islinuxabout.xyz/systray/

And lastly, @danrabbit has given a good overview talk about the problems with a system tray, and the ways we've been working to solve them within open source desktops: https://www.youtube.com/watch?v=fPFdV-Z69Lo

Native apps made for elementary OS don't use a "system tray", and cross-platform or non-native apps that run on elementary OS should not expect it to exist; instead, those apps should use standardized, cross-desktop APIs to implement their behaviors. Most apps that _can_ use a system tray icon are perfectly functional without it; however, if an app is non-functional due to the lack of a system tray, it is an application error as the system tray was explicitly never guaranteed to exist in the FreeDesktop spec, and desktops and OSes like GNOME, elementary OS, Fedora, Pop!_OS, etc. do not use a system tray.

closed 2020-05-10 by cassidyjames

AndrzejL-eu, 2020-05-11:
This very elaborate answer (which could partly be written in Chinese) does not satisfy my need for a working system tray. I do understand however that I am not getting one and that Linux community is still divided into gazillion groups and every single one is setting their standards and trying to make them "the standardized rules" while everyone is doing whatever they want and they cannot reach consensus even on the most basic things. What does that mean for a simple users like myself? It means we are not happy most of the time and we keep jumping between distros / desktop environments without ever reaching the complete happiness... 

Don't get me wrong - I am not bashing your post. Its very good. Its polite. Its very smart. What I am saying is - its difficult to be a happy Linux user.

I thank you for taking time to post a reply. You have a great day.

Kind regards.

Andrzej

SubSonicSoftware, 2023-04-23:
I used to run a couple tray icons prior to switching to Elementary/Pantheon. I'm having trouble adapting scripts that were originally cobbled together from Google searches long ago.

The icon script is:
```
#!/usr/bin/python
import os
from gi.repository import Gtk as gtk, AppIndicator3 as appindicator

def main():
  indicator = appindicator.Indicator.new("timeshifttray", "timeshift", appindicator.IndicatorCategory.APPLICATION_STATUS)
  indicator.set_status(appindicator.IndicatorStatus.ACTIVE)
  indicator.set_menu(menu())
  gtk.main()

def menu():
  menu = gtk.Menu()
  
#  command_one = gtk.MenuItem('Open Main Window')
#  command_one.connect('activate', open)
#  menu.append(command_one)

#  exittray = gtk.MenuItem('Exit Tray')
#  exittray.connect('activate', quit)
#  menu.append(exittray)
  
  menu.show_all()
  return menu
  
def open(_):
  os.system("sudo timeshift-gtk")

def quit(_):
  gtk.main_quit()

if __name__ == "__main__":
  main()

```
I've installed https://github.com/MvBonin/wingpanel-community-indicators & it appears to start without error, however the python script fails to display anything. It returns warnings I can usually ignore on KDE:
```
python /home/me/Documents/Scripts/Syncovery
/home/me/Documents/Scripts/Syncovery:3: PyGIWarning: Gtk was imported without specifying a version first. Use gi.require_version('Gtk', '3.0') before import to ensure that the right version gets loaded.
  from gi.repository import Gtk as gtk, AppIndicator3 as appindicator
/home/me/Documents/Scripts/Syncovery:3: PyGIWarning: AppIndicator3 was imported without specifying a version first. Use gi.require_version('AppIndicator3', '0.1') before import to ensure that the right version gets loaded.
  from gi.repository import Gtk as gtk, AppIndicator3 as appindicator
```
Not sure how to proceed.