elementary/settings-power #151: Move to SettingsSidebar

cassidyjames, 2020-05-14:
There are a lot of settings here (especially on laptops), and we could probably better sort them using a settings sidebar. My initial thought for grouping is something like:

- Display (brightness, auto-adjust, turn off when inactive)
- Hardware (power button and lid settings)
- Suspend (plugged in and battery)

danrabbit, 2020-05-14:
"Hardware" is a little opaque to me, but I'm not sure I have a better solution.

Something to consider is that lid settings only apply to laptops, so on tablets and desktops you'd have a sidebar pane with only one setting, which is kind of weird.

Same with suspend. You'd just have the one setting on a desktop.

cassidyjames, 2020-05-14:
ah yeah hrm