elementary/settings-power #165: Changing "Suspensiion Options" switches Desktop Theme to "Default" (White Theme)

lorencinibrdev, 2020-10-04:
**Description of the issue:**
On a fully updated "elementary OS 6.0 Next" system, and with the Dark Theme enabled system-wide, when trying to change the "Suspension Options" for the machine, on the Power Plug, the theme goes back to the Default One as soon as an option is selected on the drop-down.

**Before Change:**
[image]

**After Changing:**
[image]

**Steps to reproduce the behaviour:**
1. Activate the "Dark Theme" on Desktop Plug.
2. Go to "Power Plug"
3. Change the "Suspend when inactive for..." option to any value.
4. Theme should auto-switch to Default White One.

**Expected behaviour**
System theme should not be auto-changed.

**System Info:**
[image]

[image]

lorencinibrdev, 2020-10-12:
I've noticed the same behaviour happens on the "Date and Time" plug. Just by entering it, whole system theme goes to White:

[image]

cassidyjames, 2020-11-05:
@lorencinibrdev I can no longer reproduce either of these on a fully updated elementary OS 6 install. Are they still an issue for you?

fabiozaramella, 2020-11-06:
I can confirm that it's still happening on an updated Odin daily machine.

[image]

lorencinibrdev, 2020-11-17:
> I can confirm that it's still happening on an updated Odin daily machine.
> 
> [image]

Same here. Weird thing is that this seems to be happening on different modules. I've seen it happening on Power Plug, as well as in Date and Time plug.

hanaral, 2021-01-07:
I can't reproduce it at all, but I have my dark style preference to sunset to sunrise. Could you see if you can at all reproduce it with the latest builds of System Settings?

im-barney, 2021-05-01:
I have the same issue as described above. 
Elementary OS 6 Early Access.

tintou, 2021-05-19:
I don't have this issue here, I need more logs of what is happening.
Can you open a terminal and run `G_MESSAGES_DEBUG=all io.elementary.switchboard` then reproduce the issue and paste the output?
I'd be also interested by the logs provided by `journalctl -b -r` right after you hit the issue.

im-barney, 2021-05-23:
I went to get the logs but did an update first. Perhaps I shouldn't have?
Can't reproduce the error any more.

lorencinibrdev, 2021-09-03:
Hi all.

I'm sorry for the delay. As of the current release, i cannot reproduce the issue anymore. Can i close this?

im-barney, 2021-09-03:
I haven't experienced this since the public release. Happy to close :)

closed 2021-09-03 by meisenzahl