elementary/settings-power #166: Power Settings do not apply to the lock screen

jeregrine, 2020-10-21:
I've been struggling with my power settings, I've set it to do not standby or turn off for any reason but I'd come back to a computer desktop that's suspended.

(And it only comes out of suspend successfully 1/5 times anyways).

Turns out the fix was to disable the lock screen which I assume has power options set to standby when idle. So now I just turn my monitors off and walk away and it seems fine.

Blast-City, 2021-01-02:
Hi.

Does the following text is present on the /var/lib/AccountsService/users/your_user_name_here file.

```
[io.elementary.pantheon.AccountsService]
SleepInactiveBatteryTimeout=0
SleepInactiveBatteryType=5
SleepInactiveACTimeout=0
SleepInactiveACType=5
```

jeregrine, 2021-01-03:
Sorry I no longer have this machine available to test or provide feedback.

closed 2021-01-03 by jeregrine