elementary/settings-power #167: Add show percent in panel

danrabbit, 2020-10-28:
## Problem

Choosing to show the battery percentage in the panel isn't something you frequently reference or change, so it probably doesn't need to be in the indicator. But there's currently nowhere else to change this setting

## Proposal

Put the setting here so we can remove it from the indicator

cassidyjames, 2020-10-28:
I'm unsure if we should remove it from the indicator, but it makes sense to have this here, too, regardless. :+1:

cassidyjames, 2021-01-13:
@hanaral see: https://github.com/elementary/wingpanel-indicator-power/issues/149

andirsun, 2021-07-25:
As I implement this PR https://github.com/elementary/wingpanel-indicator-power/pull/194 
then I will migrate switch to power settings

andirsun, 2021-07-25:
@cassidyjames @danrabbit Done ! 
First merge https://github.com/elementary/switchboard-plug-power/pull/193
Then merge https://github.com/elementary/wingpanel-indicator-power/pull/195

PR #193 (2021-08-17): Show percentage switch
fixes: #167
closed 2021-08-17 by cassidyjames