elementary/settings-power #173: Disable brightness controls for external displays

ayoungethan, 2021-01-31:
As per https://github.com/elementary/wingpanel-indicator-power/issues/163

on 5.1.7

The brightness control settings are irrelevant for systems using only external displays. Suggest that both the display brightness slider and the "automatically adjust brightness" button be dimmed out/disabled when only external displays are detected, to avoid the appearance of hardware control capacity that does not exist.

This is similar, eg, to behavior that disables software control of bluetooth when no bluetooth device is connected.

cassidyjames, 2021-02-02:
Some external displays may actually support brightness controls; we should check if brightness control is supported on any enabled displays before disabling stuff.

ayoungethan, 2021-02-02:
https://en.wikipedia.org/wiki/Display_Data_Channel#DDC/CI

> a means for a computer to send commands to the monitor, as well as receive sensor data from the monitor, over a bidirectional link. 
> Despite its ubiquity in post-2016 displays, DDC/CI is not generally used by the operating system by default for brightness control on external displays.[9]

I wonder if it is a long-standing myth amongst computer users and developers that external displays do not support bi-directional communication and OS-sourced commands? Most of my searching has come up with people saying, "External displays don't support this." But if the above is true, in fact most displays now support this; it is the OS that simply doesn't make use of it!

Maccer1, 2022-05-23:
Coming back to this:

The tool: [ddccontrol](https://manpages.ubuntu.com/manpages/focal/man1/ddccontrol.1.html)

Short explanation: [how to change brightness](https://askubuntu.com/a/1181157)

To add support for new models: `LANG= LC_ALL= ddccontrol -p -c -d` and share it with ddccontrol-users@lists.sourceforge.net

This tool works but I'm unsure if they are actively developing it. Anyhow, I could change the brightness easily first time I opened it with a generic profile.

There is also [ddcutil](https://www.ddcutil.com/) but I still need to understand it.

Maccer1, 2022-05-23:
To change brightness with ddcuitil from cli:

`sudo  ddcutil detect` --> will provide list of displays and specify the ones supported.
`sudo  ddcutil getvcp known --display Y` --> will provide VCP codes of display Y.

In my case:

`VCP code 0x10 (Brightness                    ): current value =    30, max value =   100
`

To change brightness you need to change the VCP code value with:

`sudo ddcutil setvcp 10 XX`where XX is the brightness value and 10 comes from `0x10`

---

Extra info:

* You can install `i2c-tools` and add your user to the group `i2c` to do this without root

* Open Issue: Check if brightness is always VCP 0x10.

Maccer1, 2022-05-24:
Related to https://github.com/elementary/wingpanel-indicator-power/issues/41