elementary/settings-power #180: Brightness control stopped working

rigglemania, 2021-05-03:
## What Happened
When I open the Power settings, the brightness slider worked the first time, but then seemed to stop working. I tried toggling the option to auto adjust brightness but that had no effect. The function keys on my laptop also stopped working so this may be a wider system issue.  I'm not sure if this matters, but I was also running in Demo Mode as I'm wanting to wait a bit of time for some issues to be resolved before installing it on my disk.

## Expected Behavior
Brightness slider adjusts brightness

## Steps to Reproduce
1. Open System Systems
2. Open Power
3. Adjust "Display brightness" slider

## Platform Information
elementary OS 6 Early Access
Linux 5.8.0-50-generic
Lenovo Thinkpad X1 Carbon 7th Gen

markob94, 2021-05-05:
I have the same issue.

HP-ENVY-x360-Convertible-13
Quad-Core AMD Ryzen 5 3500u with radeon vega mobile gfx
Advanced Micro Devices, Inc. [AMD®/ATI] Picasso

asceel, 2021-05-17:
Same issue here but I have eOS installed; not on demo mode.
eOS 6 Early Access
Linux 5.8.0-53-generic
Acer Aspire-V5
Dual-Core Intel Core i7-4500U

Waester, 2021-08-13:
I have the same issue in Odin, worked fine in Hera.

**Platform Information**
elementary OS 6 Odin
Linux 5.11.0-25-generic
HP EliteBook 830 G5

N1sek, 2021-08-19:
I have the same issue. It happens when I try to slide the brightness control quickly.

**Device information**
elementary OS 6 Odin
Linux 5.11.0-25-generic
Quad-Core Intel® Core™ i5-8259U CPU @ 2.30GHz
Intel® Iris Plus Graphics 655

davidmhewitt, 2021-08-24:
Fixed in #202 

This will be pushed out as an update when a new release of the plug is prepared

closed 2021-08-24 by davidmhewitt