elementary/settings-power #186: Cannot resume from automatic suspend

selwynorren, 2021-07-01:
## What Happened
Once computer enter suspend mode after a period of time set in the power plug in switchboard and I try resume operations, it never resumes. The pop up for the password shows and I can enter it  successfully, but after that I just see a blank screen (On both my screens) and the mouse, but nothing else, no matter how long I wait, eventually the screen just blanks out and I can input my password again, but it never fully restores from that suspend.

I can press Control + F1 and do a lightdm restart then it all works but then I has closed all my app that were open, kind pointless then

## Expected Behavior
Should be able to enter suspend mode and fully recover form that as well.

## Steps to Reproduce
Enter system mode and see if you can recover

### Logs
Cant find where logs would be located, point me in the direction and I am happy to supply them

## Platform Information
I have latest daily snapshot with all updates install. I also have NVIDIA driver 460 installed (but the alternate way as mentioned in my bug report regarding driver install).

My graphics card is the GTX1060 with dual monitors

andirsun, 2021-08-20:
Thanks for report this, It seems a duplicate of https://github.com/elementary/gala/issues/1126.
You can close the issue.

selwynorren, 2021-08-21:
Perfect I will close the issue, thanks so much

closed 2021-08-21 by selwynorren