elementary/settings-power #200: Display brightness adjustment stops working

volfpeter, 2021-08-19:
## What Happened

Manual display brightness adjustment permanently stops working when Power settings is opened and brightness is changed using the brightness adjustment keys.

## Expected Behavior

Brightness adjustment keeps working.

## Steps to Reproduce

- Open Power settings.
- Repeatedly increase or decrease display brightness using the laptop's corresponding keys.

Brightness adjustment stops working after a relatively small number of changes (min to max and back to min should be about enough to reproduce the problem).

## Platform Information

Elementary OS 6 (not beta) - Dell XPS 9500

volfpeter, 2021-08-19:
Follow-up problem:
Once brightness adjustment breaks, if I lock the screen and log back again, I get a black screen with only the mouse pointer visible.

tomaswarynyca, 2021-08-19:
> Follow-up problem:
> Once brightness adjustment breaks, if I lock the screen and log back again, I get a black screen with only the mouse pointer visible.

@volfpeter This is a gala problem https://github.com/elementary/gala/issues/1126

andirsun, 2021-08-20:
Duplicate #180, please close de issue.

closed 2021-08-20 by volfpeter