elementary/settings-power #204: power-profiles-daemon integration

jhaygood86, 2021-09-10:
## Problem

elementary OS should allow native management of "power profiles" to either improve performance, battery life, or use a balanced mode.

## Proposal
Bastien Nocera's excellent power-profiles-daemon provides a native D-BUS API that allows managing these profiles in a cross-hardware fashion, exposing 3 modes: balanced, power-saver, performance. GNOME already provides integration for it.

The daemon would need to be packaged for focal. There's some minor patches that would be needed since its intended for newer libgudev than what focal has packaged, but the functionality is available with the version focal packages in a slightly different way.

Note that Fedora 35 is going to package it and integrate it with GNOME Control Center.

This would be an excellent feature to ship in a feature update to elementary OS 6 (6.1?)

## Prior Art
Bastien Nocera documents a lot of prior art on the gitlab page at https://gitlab.freedesktop.org/hadess/power-profiles-daemon

## Side Note
I would be interested in helping integrate this, but I don't know the process to package new dependencies for elementary OS

SubhadeepJasu, 2021-09-10:
WIP #192

danrabbit, 2022-01-10:
Fixed in #192

closed 2022-01-10 by danrabbit