elementary/settings-power #215: Screen is dark after wakeup from suspend

2whlrider, 2022-03-13:
### What Happened?

6.1 Jolnir Power Button settings are 15min/Suspend/15min. My desktop successfully suspends and successfully wakes up on key press but my monitor remains dark. If I shut off my monitor and restart I get the elementary login screen. The monitor is apparently not getting the wakeup signal it needs to resume after suspend.

### Steps to Reproduce

Power Settings: Turn off display when active: 15 minutes (same non-wake-up behavior if set to "never") 
Power Button: Suspend
Suspend when inactive  for: 15 minutes

### Expected Behavior

I expect the monitor to wake-up when the OS resumes from suspension.

### OS Version

6.x (Odin)

### Software Version

Latest release (I have run all updates)

### Hardware Info

Asrock B450 Pro 4 motherboard
AMD Ryzen 2600
Powercolor Radian RX 580
Lenovo Thinkvision 24" Monitor

justinmichaelvieira, 2023-01-13:
I am also experiencing this issue, on elementary OS 6.1 Jólnir

Hardware info:
Huawei Matebook Pro (2020)
Quad-Core Intel® Core™ i7-8550U CPU @ 1.80GHz
Intel® UHD Graphics 620 (Kabylake GT2)
NVIDIA Corporation GP108M [GeForce MX150]
16GB memory

When the laptop wakes from Suspend, screen brightness is consistently at the lowest possible setting. Regardless of what screen brightness was set to, before the laptop was suspended.

baloghszilveszter, 2023-02-05:
This solved the black screen problem for me:

I added this line to **/etc/default/grub**

> GRUB_CMDLINE_LINUX="intel_idle.max_cstate=1 i915.enable_dc=0"

then 

> sudo update-grub &&  sudo reboot

Hardware info:

Dell Latitude 5590
CPU: Intel i7-8650U (8) @ 4.200GHz
GPU: Intel UHD Graphics 620
Memory: 16GB