elementary/settings-power #235: notification about suspending after 5 minutess of inactivity contradicts my settings

janxkoci, 2023-06-02:
### What Happened?

I'm getting a lot of notifications warning me about suspending my laptop only after a few (~5) minutes of inactivity, even though my settings look like this:

[image]

The notifications are very disruptive when I try to focus on reading or similar. How do I stop them please?

### Steps to Reproduce

1. Open reading material, e.g. a science paper.
2. Read for a few minutes without touching a mouse.
3. Get annoyed by a notification that does not respect your settings.

### Expected Behavior

If there should be a warning notification, then it should come around the time I specified.

### OS Version

7.x (Horus)

### Software Version

Latest release (I have run all updates)

### Hardware Info

- Slimbook Essential 14.
- elementary OS 7
- [slimbook battery](https://github.com/Slimbook-Team/slimbookbattery) (from PPA)
  - this is because the battery profiles in elementary OS are not sufficient

janxkoci, 2023-08-01:
# Follow-up
A few days ago I left my laptop awake in my office while being away for a longer time. The laptop **didn't actually go to "sleep"** as I set it to do (and as the annoying notification warns). Instead,

- only the screen was off (black)
- the laptop stayed awake (power LED was ON, not blinking, and the fan was spinning)

I tried to bring it back by pressing the power button or wake-up key on the keyboard, then other keys as well as touchpad, but nothing worked. In the end **I had to force shutdown** by holding the power button for several seconds, losing my session.

This is not cool. :frowning_face: 

## Few notes

The notification in question has the following features:

- it shows a warning triangle icon with a `!` mark and a battery emblem icon in the lower-right corner;
- no such icon shows up in the swithboard plug, so I cannot change its settings (disable sound for example);
- it persists visible on the screen until I move my mouse or otherwise interact with the system, at which point it disappears;
- for the above reason it rarely appears in the notification widget in wingpanel, with one exception - the incident described above, after which I cold booted to my system and found the notification there.

The notification comes every few minutes when I don't interact with the OS, most often when I'm reading some science paper. Since I cannot disable the notification or its sound, it's super-distractive at times when I need to focus. I really wish to never see it, as it's pointless and not doing anything useful (the opposite in fact).

**Is there a way to figure out which process sends the notification?** Probably post-fact in some logs, as any interaction makes it go away.

lenemter, 2024-01-26:
Fixed in https://github.com/elementary/notifications/pull/200

closed 2024-01-26 by lenemter