elementary/settings-power #237: power settings. power consumption mode

pennywise322, 2023-06-29:
### Problem

the problem is that in the settings it is not possible to set a different default energy consumption mode for connected to the mains and energy consumption from the accumulator.

### Proposal

I have a laptop and it would be nice if my laptop would work at full power when I am plugged in. but when I disconnect from the mains I want it to work in saving mode or balance. it would also be good to add power mode management to the system indicator battery icon on the panel. so that you don't have to go into the settings every time you want to change the electricity consumption mode, and you could do it easily and conveniently

jeremypw, 2023-07-12:
I am sure this can be fixed but there are a limited number of devs working on a lot of different things.  Please be patient.

lenemter, 2024-01-27:
Closing as duplicate of https://github.com/elementary/switchboard-plug-power/issues/227

closed 2024-01-27 by lenemter