elementary/settings-power #270: Battery Charge Limit

myhousece, 2024-08-10:
### Problem

I saw that there are already requests for this feature here, but I would like to make this new addition, because I think it is extremely important for notebooks, especially those that have this feature, like my notebook "ROG-Zephyrus-G15-GA503RM".

That would be to be able to limit the battery charge to better protect its health.

### Proposal

I have already searched in several distributions, environments, Gnome, KDE and I don't see it. There is even a native option in KDE Plasma, but when I activate it, it only works while the notebook is on and if I turn it off or restart it, it goes back to 100% charge. It does not maintain the 60%, 80% configuration. And I believe that this feature is extremely important, it would be a great joy to be able to enjoy this feature in Elementary OS, a distro that I have a lot of affection for.
[image]

lenemter, 2024-08-27:
Upower recently added that functionality into their dbus interface ([here](https://gitlab.freedesktop.org/upower/upower/-/releases/v1.90.5)) once ubuntu 24.04 gets upower 1.90.5 we can easily implement that feature.

danirabbit, 2026-09-08:
For future reference, here's some docs: https://upower.freedesktop.org/docs/Device.html

How it looks on macOS:

[image]

Windows the design changes by laptop manufacturer (wtf is windows) but here's Asus:

[image]

GNOME:

[image]